micropython/micropython #2845: Preparing for next release, v1.9

dpgeorge, 2017-02-07:
There have been some breaking changes since the last version (notably refactoring of VFS sub-system), and I want to make a few more user-facing changes to stmhal (at least) so the next version should be v1.9.  Since there's still a bit of work to do the release won't be at the usual time (around first week of the month, ie now) but instead likely delayed till the next month.

pfalcon, 2017-02-24:
* https://github.com/micropython/micropython/issues/2850 "esp8266: sys.path broken"

pfalcon, 2017-04-25:
Above is the updated list of "blockers" for the release.

Besides long time w/o a release, which is the most pressing factor, various other (smaller) issues pop up, e.g. zephyr port needs to be changed incompatibly for Zephyr 1.7 (current release) and Zephyr 1.8 (next release), would be nice to have uPy version matching current Z release.

dpgeorge, 2017-04-26:
I added a few more items to the list.  The "long lines" one is a regression that needs to be fixed (at minimum the change should be reverted).  The "mpy dependency" one would be good to get working because this release will have updated bytecode and will need the cross compiler to be rebuilt.

pfalcon, 2017-05-01:
esp-extra-scripts was rebased and uasyncio was added there.

pfalcon, 2017-05-10:
With addition of uasyncio, esp-extra-scripts overflows flash by a lot.

dpgeorge, 2017-05-11:
> With addition of uasyncio, esp-extra-scripts overflows flash by a lot.

It looks like there are duplicate symlinks in scripts/ and modules/ (http* and example*, ssd1306, 7 duplicates in total).  Removing the duplicates takes the overflow from 20680 down to 8748.  Moving the remaining (non dup) files from scripts/ to modules/ gets the overflow down to 8268 (ie using frozen bytecode instead of frozen text saves about 500 bytes).

pfalcon, 2017-05-11:
> 8748. Moving the remaining (non dup) files from scripts/ to modules/ gets the overflow down to 8268 (ie using frozen bytecode instead of frozen text saves about 500 bytes).

Yeah, and that's very little savings :-I.

And I was about to push the change from https://github.com/micropython/micropython/pull/2835 to close it, and this case shows again that rigorously saving every byte is very important in our business.

So, we need to grow firmware size, grow by large - 3 sectors. We could grow just esp_extra_scripts, but that's not very practical for workflow on our side, requiring flash erase switching between master and esp_extra_scripts. And that's not the last increase, and the we need to decide fate of 1MB Flash support - filesystem size there grows more and more non-useful. We can decide that it's ok, and for real use, 4MB is required, and people with 1MB, who want to get the most out of it, will need to do custom configuration, just as for 512KB.

dpgeorge, 2017-05-11:
After moving all scripts/ to modules/ one can disable MICROPY_MODULE_FROZEN_STR completely and that saves another 492 bytes.

We can remove the example scripts, but I guess in the long run we need to increase the FW size.

> and the we need to decide fate of 1MB Flash support - filesystem size there grows more and more non-useful.

I think 384k of filesystem is still usable, and even 256k is a lot better than nothing (for 1MB modules).  384k allows us to increase firmware size by 64k.

> We can decide that it's ok, and for real use, 4MB is required,

Going down this path (eg starting the FS at 1MB position) might allow to have the same flash layout for OTA and non-OTA builds.

pfalcon, 2017-05-11:
> After moving all scripts/ to modules/ one can disable MICROPY_MODULE_FROZEN_STR completely and that saves another 492 bytes.

Well, no, resource_stream() which requires it was specifically added to allow real-world frozen apps. But if we had common lookup code for both frozen types, we'd probably saved 200 bytes ;-).

> Going down this path (eg starting the FS at 1MB position) might allow to have the same flash layout for OTA and non-OTA builds.

That's not what I suggest, just that 1MB build will have a "toy" filesystem.

> I think 384k of filesystem is still usable, and even 256k is a lot better than nothing (for 1MB modules). 384k allows us to increase firmware size by 64k.

Ok, so let's grow firmware by some size which would allow not to grow it very soon, e.g. by 16K.

dpgeorge, 2017-05-11:
Note that compiling .mpy files with -O3 (full optimisation, no line numbers stored) saves 993 bytes.

pfalcon, 2017-05-11:
Yeah, I tried that too. uasyncio stayed at 5xxx bytes.

pfalcon, 2017-05-11:
Actually, let's get realistic and grow it by 32k: https://github.com/micropython/micropython/pull/3081

dpgeorge, 2017-05-13:
Ok, it's going to take some time to write the release notes for 1.9 because there were heaps of changes since 1.8.7.  In particular I want to make the effort to point out the following things in the notes:
- any incompatibilities from a user's point of view (eg bytecode changed)
- any incompatibilities from a port developer/maintainer's point of view (eg some signatures of "public" C functions)
- code size change (since it changed a lot)
- major new features

The change in code size so far since 1.8.7 is:
```
   bare-arm: -1096
    minimal: -1424
   unix x64: -1155
unix nanbox: -11506
     stmhal: +16988
     cc3200: +1072
    esp8266: +3012
```
Remember that quite a few features were added to the core (eg new bytecodes) so a reduction in code size by 1k is pretty nice.  nanbox dropped by a lot due to conversion of mp_uint_t to size_t (and also performance increased a fair bit).  stmhal grew due to new frozen bytecode.

If anyone things of things to mention in the release notes, please mention them here.

pfalcon, 2017-05-13:
@dpgeorge : Please don't forget that https://github.com/micropython/micropython/pull/3025 is on critical path to the release.

pfalcon, 2017-05-13:
>  I want to make the effort to point out the following things in the notes

My suggestion would be to take it easy. As you say, it would be quite an effort to do this release "as usual", so no need to perfectalize it beyond that. Soon, very soon we'll finish the core of MicroPython and then we'll be able to pay attention to all those "stats" ;-).

pfalcon, 2017-05-21:
We now have all important tickets resolved, so I guess it would be nice to cut the release, and then start to prepare to 1.9.1 bugfix release ;-).

peterhinch, 2017-05-22:
If uasyncio is to be a frozen module you might want to fix https://github.com/micropython/micropython-lib/issues/179

dpgeorge, 2017-05-23:
There're still a couple of minor make-related things to fix: #2959 and #3099

dpgeorge, 2017-05-25:
Ok, during extensive testing some bugs/regressions were found in the vm.  They have now been fixed (see 68e71eacb8f5e67f9765ccd8639ca359bc3417de, 04d05db27ec4a160dcb3d3ff2934184fef455dbf and 8f064e469d551ab51a960cebf011e34af5e7fa58).

Patches for the final 2 issues mentioned above have been posted.

dpgeorge, 2017-05-26:
Everything now seems to be finished for the v1.9 release.  Some testing will be done then the tag should be applied within the next couple of days.

pfalcon, 2017-05-26:
@dpgeorge : Great, thanks! Can you please give more specific ETA, e.g. would you make the release on the weekend?

dpgeorge, 2017-05-26:
> Can you please give more specific ETA

I will try to do it ASAP, maybe in the next few hours if there are no problems.  Otherwise I can aim to have it done by UTC23:00 Sat 27th May.  Are there things you want to add?

pfalcon, 2017-05-26:
There's always something to clean up ;-) (like, warning in Zephyr port). Nope, it's not too important, if you're set to do it soon, please do.

dpgeorge, 2017-05-26:
@pfalcon are you aware that uasyncio still has issues on esp8266?  Eg example_http_client.py (from micropython-lib/uasyncio) doesn't work due to the bug described in #2930.  example_http_server.py does work though.

pfalcon, 2017-05-26:
I may imagine there're issues. I didn't have time to test everything, and short on time now, I don't think it's worth delaying the release.

dpgeorge, 2017-05-26:
That bug in particular *should* be a simple fix, but I agree it's best to leave it until after v1.9.

dpgeorge, 2017-05-26:
v1.9 was tagged in 825460a093a6bcd8fb79119b5f6ee8408f63603b

closed 2017-05-26 by dpgeorge

dpgeorge, 2017-05-26:
esp-extra-scripts was rebased on to v1.9

pfalcon, 2017-05-27:
Great, thanks @dpgeorge !