micropython/micropython #2846: RFC: Soft IRQ handling via micropython.schedule

dpgeorge, 2017-02-07:
Following on from the proof-of-concept work done in #2081 (adding support for soft interrupt handlers), my proposal for a generic way to do soft IRQ handling is to add a function:
```python
micropython.schedule(func, arg)
```
This function has the following behaviour:
- It "schedules" the function/callback `func` to be called at a later date, with the argument `arg`, ie `func(arg)`.
- `func` will only be called when the heap is unlocked and "atomic" operations (eg list insert) are complete.
- schedule will try to call `func` as soon as possible.
- If schedule is called in normal Python code (eg with heap unlocked) then schedule can chose to execute `func` immediately.
- More than one `func` can be in the schedule queue at once.  Maximum size of queue is port-dependent.  If there's no room then schedule raises an exception.
- New `func`'s are placed at the end of the queue (priorities may be added in the future).

With the above function (and the VM's ability to run scheduled functions when appropriate), ports that already have hard IRQs can immediately implement soft IRQs, eg:
```python
def hard_irq(arg):
    micropython.schudule(soft_irq, arg)
```
But of course micropython.schedule will be used internally by ports to implement soft IRQs directly, eg `timer.irq(func, arg, hard=False)` (soft will be the default).  The reason to have micropython.schedule exposed to the user is so they can do a combination of hard and soft IRQs if needed.  It also allows the unix port to implement soft IRQs.

pfalcon, 2017-02-07:
Ok, so if ability to specify soft irq directly via .irq() method is still there, I guess it sounds good.

I also mentioned previously that uasyncio and this mechanism either try to approach a similar goal from different ends, or just follow parallel approaches. It would be nice if implementation went with that in mind (so, it was clear how they differ, perhaps trying to avoid duplicating too much functionality; I don't think there will be much to reuse between 2 impls).

pfalcon, 2017-02-07:
> schedule will try to call func as soon as possible.

Also, I'm not sure if it makes sense to promise that. As discussed previously, one (obvious?) way to implement this processing is via existing MICROPY_VM_HOOK_LOOP, perhaps with a divisor like esp8266 does.

dhylands, 2017-02-07:
I can see a couple different types of priority:
1 - How soon it will execute (alluding to @pfalcon's comment and perhaps being able to set the divisor)
2 - How it will compete with already scheduled functions (I think that this is the priority that @dpgeorge is referring to)

If the soft-irqs are being used as a hard-irq replacement, the having how soon be "as soon as possible" makes sense to me. For these situations you really don't want to be introducing additional delay. If you're responding to hardware events, every delay means that you're effectively slowing down the maximum rate of events you can respond to.

There are lots of other usage scenarios where adding a small delay is perfectly acceptable. If you're debouncing a push button, then the fact you don't execute for a millisecond or two is typically inconsequential. If you start to delay by 100's of milliseconds then that becomes noticeable to a user.

peterhinch, 2017-02-07:
One use case is where a hard ISR acquires data to a pre-allocated buffer or queue, handles any hardware protocol, then schedules a soft ISR to process it. To be useful the soft ISR will need to be called more quickly than, say, having an asyncio coroutine poll a flag set by the ISR. Also more promptly than an asyncio coro scheduled by the planned IORead/uselect mechanism.

There remains the hazard of a subsequent hard IRQ occurring before the soft ISR has started or while it is in progress. But that's inescapable - users will need to code accordingly.

dpgeorge, 2017-02-08:
> also mentioned previously that uasyncio and this mechanism either try to approach a similar goal from different ends, or just follow parallel approaches.

I really don't think there's any useful overlap with uasyncio and soft IRQs: the latter has no concept of time scheduling, nor of generators.  It expects the function to run to completion.

> schedule will try to call func as soon as possible.
> 
> Also, I'm not sure if it makes sense to promise that.

There needs to be some clause like that in the specification, otherwise the simplest implementation is to never run the scheduled function :)

> As discussed previously, one (obvious?) way to implement this processing is via existing MICROPY_VM_HOOK_LOOP, perhaps with a divisor like esp8266 does.

The approach implemented in #2081 used the "pending exception" check to check for soft irqs: this is both more efficient (the check is already done by the VM so it adds no overhead) and lowers the latency (they check is done every loop opcode).  So I'd try to make it work this way.

Despite the simplicity of micropython.schedule, there are lots of details to work out, eg:
- If one calls time.sleep(10) and irqs happen while sleeping, one would expect all soft irqs to run and for the sleep to return after 10 seconds.  Thus, the sleep loop must check for pending scheduled functions and run them, while still keeping track of time.  Similar story for other blocking functions like select/poll.
- With multithreading enabled, which thread runs the scheduled funcs?  (I'd say which ever thread is currently running.)  Can the VM switch to other threads while a scheduled func is running?  (I'd say no.)

I also considered having a timestamp to go with the scheduled func (the stamp would be at the time of scheduling).  Arguably you'd want microsecond precision, so you'd take the value from time.ticks_us() as the stamp.  This could be passed as an argument to the callback, eg `def soft_irq_handler(arg, timestamp): ...`.  You could then use time.ticks_diff() to see how late the soft irq is called from the actual scheduling.  With this addition, one would want to follow suit with hard irqs and add such an argument.  I'd be reluctant to add such a complication.

pfalcon, 2017-02-08:
> If one calls time.sleep(10) and irqs happen while sleeping, one would expect all soft irqs to run and for the sleep to return after 10 seconds. Thus, the sleep loop must check for pending scheduled [...]

That's why I keep repeating the request to look into doing it with MICROPY_VM_HOOK_LOOP - the mechanism which is needed to be implemented, it is (very) similar to MICROPY_VM_HOOK_LOOP and other "event polling" infra we developed for ESP8266. But there it works for "underlying system", and here we're talking about reusing it for uPy interpreter itself.

pfalcon, 2017-02-08:
> I really don't think there's any useful overlap with uasyncio and soft IRQs: the latter has no concept of time scheduling, nor of generators. It expects the function to run to completion.

Yes, there's one thing in common: they offer scheduling (which requires queue, and thus heavy on memory). So, if there's a scheduling implementation with advanced features you mentioned, that it would be good design approach for an alternative to keep it as simple as possible, also saving on resources trying to reuse as much as possible (not between 2 scheduling impls, in general).

dpgeorge, 2017-02-08:
> Yes, there's one thing in common: they offer scheduling

With uasyncio, the scheduled coro needs to return something to the entity that scheduled it.  With soft irqs, the return value of the scheduled function is lost, there is no one to receive it (because it's essentially run asynchronously).  And coros are scheduled to run "at some event", eg timeout or file becoming readable, so all these possibilities would need to be hard coded in micropython.schedule.  I don't see how it would work.

pfalcon, 2017-02-08:
>  I don't see how it would work.

I'm sorry if I didn't make it clear that I didn't expect it (whatever it is) to "work", just suggested that existence of one scheduling mechanism may affect *requirements* and *design* of another scheduling method. That was a suggestion in the hope to give it good direction, not trying to turn it somewhere or criticizing your ideas (if it looked like that).

hoihu, 2017-02-08:
I'm looking forward to this!

tannewt, 2017-02-20:
How does this compare to [CPython's `sched` module](https://docs.python.org/3/library/sched.html)? Would it make sense to provide that API or a different CPython API?

dhylands, 2017-02-20:
I think that the only similarity is in the name.

dpgeorge, 2017-02-21:
> How does this compare to CPython's sched module?

@tannewt it's good to think in this direction (reusing CPython modules) but in this case (as @dhylands says) there's almost no similarity with the sched module.  That module is for doing things like simulations which require adding events to a queue and running them in a loop.  In our case we are trying to schedule something to happen ASAP, and the entity that actually runs the scheduled functions is the VM itself, as opposed to someone calling sched.run() from within Python.

tannewt, 2017-02-21:
Yup, I'm thinking about how its done already in CPython. Maybe [`concurrent.futures`](https://docs.python.org/3/library/concurrent.futures.html) is more similar. It doesn't require a separate run call and the async functions happen soon after they are submitted.

dpgeorge, 2017-02-22:
> Maybe concurrent.futures is more similar.

Yes it does look similar!  But it requires you to return a futures object upon scheduling the function, which uses precious RAM (and remember that the scheduling function is often called from within an IRQ so cannot allocate any memory).

cversek, 2017-02-25:
I'm curious as to what kind of effort it would take to get the micropython scheduler to work on the ESP32 port?  I tried pulling these commits into [my fork of micropython-esp32](https://github.com/open-eio/micropython-esp32/commits/esp32)  and started tinkering to see if I could get `micropython.schedule_exc` to work.  Since I have no clue what I'm doing, I succeeded only in causing `Guru Meditation Errors of type LoadProhibited occured on core 0` (haha), when adding the `mp_handle_pending` calls to `MICROPY_EVENT_POLL_HOOK` and inserting that into the end `mp_hal_stdin_rx_chr`.  My guess is that the RTOS environment is somehow confounding this approach?

dpgeorge, 2017-02-27:
> My guess is that the RTOS environment is somehow confounding this approach?

Probably.  It will require some looking into, probably after this PR is merged.

dpgeorge, 2017-06-01:
PR #2878 was merged for this, docs added in d007351b3377f97316c4af2adcec164fb5e56dca

closed 2017-06-01 by dpgeorge