micropython/micropython #2847: ESP8266 - pin.irq().disable() is not available?

larsks, 2017-02-07:
The documentation suggests that one can disable an interrupt handler on a pin like this:

    mypin.irq().disable()

But at least using `v1.8.7-200-g21f0852` this method isn't available:

```
>>> p.irq().disable()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'IRQ' object has no attribute 'disable'
```

larsks, 2017-02-07:
Apparently, I was looking at the documentation for the wipy board, rather than for the esp8266 (and the latter does not describe a `.disable` method).  Sorry for the noise.

closed 2017-02-07 by larsks

dpgeorge, 2017-02-08:
These inconsistencies will be cleaned up, hopefully sooner rather than later :)

borut, 2017-06-05:
p = Pin(14, Pin.IN, Pin.PULL_UP,value=1)
p.irq(trigger=Pin.IRQ_FALLING,handler=callback)

p.Pin(14,Pin.IN)  # disables irq

Arduous, 2018-03-17:
I am using micropython 1.9.3 on a nodemcu v3.The solution proposed by @borut did not disable the interrupts in my case
p.irq(trigger=0,handler=callback) seems to properly disable the interruptions