micropython/micropython #2850: esp8266: sys.path broken

pfalcon, 2017-02-07:
In today's built binary:

~~~
>>> import sys
>>> sys.path
['', '/flash/lib', '/flash']
~~~

I may imagine this is related to OOVFS refactor.

This breaks upip.

pfalcon, 2017-02-07:
Also:
~~~
>>> os.mkdir("/lib")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
>>> os.mkdir("lib")
>>> os.listdir("/lib")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
>>> os.listdir("lib")
[]
~~~

pfalcon, 2017-02-07:
(It's still on my TODO to see how mount table was implemented in new code...)

kfricke, 2017-02-08:
Out of curiosity... How should sys.path look like? I have not found hints regarding this [in the documentation](http://docs.micropython.org/en/latest/esp8266/library/sys.html?highlight=sys.path#sys.path).
It seems consistent on ESP8266 and pyboard (just as described by you). '/' is readonly and '/flash' is writeable. Which is the behavior I read about somewhere recently, but can not recall where.

dpgeorge, 2017-02-08:
The new path matches all other ports, including unix.

The reason upip works on unix (even though it tries to mkdir /home) is that if you try to create a directory in a place that you don't have permission (eg root dir), if the dir already exists then you get an EEXISTS error instead of an EPERM error (or otherwise).

So, one could argue that the VFS system should work the same: it should return EEXISTS if one tries to mkdir the path of an already-mouted device.  This would fix upip.

pfalcon, 2017-02-08:
> The new path matches all other ports, including unix.

?? What exactly do you mean by that? That unix port has non-existing `/flash/*` components in its sys.path? No, it doesn't. It was also one of the cleanups of recent esp8266 that it mounted its filesystem on the root path following well-known conventions how filesystems work (they start at root). So, if we're back to oversimplification of "other ports" with something like "root is the place to mount filesystem [and nothing else to do], and all filesystems should be mounted on the 1st level of root [and nowhere else]" or something, then we have a [serious] regression, so please open a ticket to discuss how this decision happened to be done.

> So, one could argue that the VFS system should work the same: it should return EEXISTS if one tries to mkdir the path of an already-mouted device. This would fix upip.

Yes, so this is second issue.

pfalcon, 2017-02-08:
> The new path matches all other ports, including unix.

Ok, it's clear that everyone looks at the issue from their side, so this report is about:

1. sys.path turning from expected `['', '/lib', '/']` to `['', '/flash/lib', '/flash']`, suspectedly containing non-existing paths.

2. As investigation proceeds, regression in default filesystem mounting, which is supposed to be mounted at `/` for esp8266 and other new ports, and now suspectedly is mounted at `/flash`.

pfalcon, 2017-02-08:
(Older esp8266 FS setup code might have leaked/stuck "/flash" somewhere due to known issues with the implementation, but the intention was always to mount it at root).

kfricke, 2017-02-08:
We, the users want a decision for a design to be shared across all microcontroller ports. There are currently broken situations with the different tools used to manage applications and deployments described on the forum.

dpgeorge, 2017-02-08:
> ?? What exactly do you mean by that?

The unix port has ['', '~./micropython/lib'], and so lib lives under your personal directory.  'flash' on esp8266 (and pyboard etc) mirrors your home dir.

> root is the place to mount filesystem [and nothing else to do], and all filesystems should be mounted on the 1st level of root [and nowhere else]

Yes, this was the plan all along, to make it the same as existing stmhal.  It's simple and works well.

pfalcon, 2017-02-08:
> Yes, this was the plan all along, to make it the same as existing stmhal. It's simple and works well.

Then we have another "d'oh" moment where the picture seemed clear but turned out we see it differently. Why do you think myself, who started OO VFS refactor, did it like it was for esp8266? Because I considered that the "plan all along" to make it not "simple" but "right, general, and give least user surprise". So, please open a ticket to discuss mounting support for OO VFSes.

dpgeorge, 2017-02-08:
> Why do you think myself, who started OO VFS refactor, did it like it was for esp8266?

Because that was the only way it would work with the existing fatfs library.

> So, please open a ticket to discuss mounting support for OO VFSes.

All I see is that going to another long winded discussion without concensus.  And what exactly would the discussion be about?  The way the mounting is implemented, or the way the user API works?  Or would it be about the exact names and locations of default mount points?  As I see it, stmhal's behaviour is the way to go and other (bare metal) ports can now follow suit.

pfalcon, 2017-02-08:
> Because that was the only way it would work with the existing fatfs library.

No, because that was of the requirements behind the refactor - to make VFS which works according to well-known POSIX filesystem principles (and literally all FSes nowadays work follow that paradigm). And per it, if there's a one thing you can assume about (overall) filesystem, is that it exists at `/`. (Hardly can assume anything else, e.g. writability). 

> All I see is that going to another long winded discussion without concensus. 

Again, I'm sad to hear such statements from the project BDFL. Somehow, we worked for years with that process very productively, and suddenly - the above. As for long-winded discussions without consensus, the remedy for them is well known - discussions should be focused (starting from being held in proper, not random, places), and participants should be focused and making the matter better and finding a solution. For when this stops to work like that by itself, the rules for discussions and decisions should b established.

> And what exactly would the discussion be about? The way the mounting is implemented, or the way the user API works? Or would it be about the exact names and locations of default mount points?

Mounting principles, implementation behind them, exact names and locations.

> As I see it, stmhal's behaviour is the way to go and other (bare metal) ports can now follow suit.

There's an alternative opinion that a way established with esp8266 port is better. No talking about ports with a legacy baggage, but if they're brought into picture, they can follow the suite of having a normal root filesystem like the most filesystems in a real word have, not something hanging in a thin air (follow in backwards-compatible manner if they so desire).

dpgeorge, 2017-02-09:
> And per it, if there's a one thing you can assume about (overall) filesystem, is that it exists at /. (Hardly can assume anything else, e.g. writability).

That's how it works now.  If you enabled MICROPY_VFS then you get a root filesystem.  You can cd to it and list it.

dpgeorge, 2017-02-09:
In 8f1c6d952ac695bc41afe8965aa79fbabcc6bcc1 the issue of mkdir'ing an existing mount point is fixed so that an appropriate OSError is raised.  upip now works again on esp8266.

pfalcon, 2017-02-10:
Nice, thanks! Now just need to fix user-visible regressions introduced in f9ecd484bbd698c0f85a3f40f391afeb01940052, which, as far as I can tell, wasn't submitted for review, even though it introduces the user-visible changes, and with your (over)cautious stance on changing things which aren't (not supposed to be) user-visible, as e.g. presented here: https://github.com/micropython/micropython/issues/1550#issuecomment-276290779 .

I understand it may take you more time as you work on other things, and this more time can also be used to answer yourself few simple questions: Don't you want fliesystem organization of the machine you're sitting at to be changed e.g. from `/bin` to `/flash/bin`, `/lib` to `/flash/lib`, etc? Maybe should apply to change LFS to do everything against `/flash`? Well, distros now drop it (but not spirit of it!), so maybe just make a new slash-flash-FS and everyone will rush to use it? Or maybe it's easier after all to just follow simple and well-known world model where you have a real filesystem on `/` as the most basic thing? Also: was it worth the trouble to implement a real VFS system, like some big OS, and then fall short at oversimplistic mount table implementation? And as before, this discussion would rather happen in a dedicated ticket.

pfalcon, 2017-02-10:
P.S. What's the concern anyway? listdir()? But listdir isn't a primary interface to this operation, ilistdir_ex() is, that's what VFS should implement and listdir() impl is completely VFS-independent. And this was discussed (definitely) and agreed on (in my list).

dpgeorge, 2017-05-06:
This is now fixed, esp8266's internal flash FS is mounted at /.

closed 2017-05-06 by dpgeorge