micropython/micropython #2852: L476-DISCO: dir() hard fault

stiege, 2017-02-08:
Seems like any use of dir() or tab completion for the L476-DISCO port results in a hard fault.

```quote
MicroPython v1.8.7 on 2017-02-08; L476-DISCO with STM32L476
Type "help()" for more information.
>>> dir()
[
FATAL ERROR:
HardFault
```

dpgeorge, 2017-02-08:
It works for me, on both v1.8.7 and latest master.  Please try latest master, and please erase flash completely so that new FS is built.

stiege, 2017-02-08:
Roger; I used dfu-util and the builtin bootloader rather than the on-board st-link. Not sure of the dfu-util erase command so will install st-link drivers and try that way.

You're probably right regarding the file system; although a file system comes up when I plug the device in, it's the ST one that includes mbed.htm and details.txt. I thought this was unrelated to the REPL issue.

dpgeorge, 2017-02-08:
You can program it in 2 ways: either via st-link (mini USB connector), or via DFU (micro USB connector).  There will also be 2 filesystems and 2 REPLs: one from the st-link and one from uPy on the micro USB connector.

stiege, 2017-02-08:
Thanks @dpgeorge. Erasing fixed the issue.

closed 2017-02-08 by stiege