micropython/micropython #2853: Add functions to directly write to internal flash

ryannathans, 2017-02-08:
Had a look but couldn't see an existing method for this. 

We should have functions for direct read, write and sector erase for access to internal flash. Are there any existing (undocumented?) functions to call for this other than doing access word by word through stm32 module? 

Perhaps also allowing multiple filesystem partitions in internal flash too? That would be a separate issue though.

dpgeorge, 2017-02-08:
> Had a look but couldn't see an existing method for this.

Please see pyb.Flash(): it exposes the flash as 512 byte blocks, and has readblocks and writeblocks.

> Perhaps also allowing multiple filesystem partitions in internal flash too?

That's on the roadmap, see also #2730.

ryannathans, 2017-02-08:
@dpgeorge the intention is to read and write to flash outside of the filesystem which doesn't seem possible with readblocks/writeblocks

dpgeorge, 2017-02-08:
> the intention is to read and write to flash outside of the filesystem

Ok, well, pyb.Flash() doesn't allow you to do that.

Sector sizes are large and not uniform, and I don't really want to encourage direct acces to the sectors on pyboard, for a few reasons: the firmware may grow to the point where it uses all of them; any spare sectors could be useful for implementing features like dynamic frozen bytecode.

ryannathans, 2017-02-08:
On the chip I'm working with there is 2M of flash, where three quarters is unused even after freezing in the whole application. I wish to store configuration values in one or two unused sectors that do not get wiped when reflashing micropython or the filesystem containing main.py. 

Is the best solution then to write a python library that uses the stm32 module?

dpgeorge, 2017-02-08:
> Is the best solution then to write a python library that uses the stm32 module?

You could do that, it would probably perform ok, and would be relatively portable.

Otherwise, look in stmhal/flash.c and use the flash_erase and flash_write functions (or expose them on the python side).

pfalcon, 2017-05-06:
So, one way to standardize on this would be to expose Flash as a stream interface object. That for example would allow to use it without FS. But that would make it not thread safe if accessed concurrently (because of the current position pointer), so would be not good for FS usage. So, catch 22.

dpgeorge, 2020-06-09:
I'll close this discussion due to inactivity.

For future reference, `ports/stm32/mboot/fwupdate.py` contains pure Python code to erase and write flash sectors for F7 series MCUs.

closed 2020-06-09 by dpgeorge