micropython/micropython #2854: stmhal alt function open drain doesn't work

chrismas9, 2017-02-08:
It appears that Open Drain doesn't work for Alt Functions (eg PWM) for pyb or machine. In both cases a high level drives the pin to VDD (eg 3.3V) irrespective of external connection.

Use case: I like to drive PWM LEDs from 5V using an open drain output. This way the PWM noise isn't coupled onto the ADC reference (3.3V).

Symptom: When the PWM duty cycle is set to 100% the LED glows dimly. This is because it has 1.7V across it (A=5V, K=3.3V). When the pin is configured as open drain GPIO the LED goes out. The LED is active low so 100% dc should be off.

Test hardware: pyboard v1.0 running pybv10-20170208-v1.8.7-202-gaf622eb.dfu
Connect a pullup resistor from X3 to 5V and a voltmeter from X3 to GND.
Run the two scripts below, one for pyb and one for machine.

When the pin is low the voltmeter will read approx. 0V.
When the pin is high the voltmeter will read approx. 5V if open drain is working or 3.3V if it is not working.

```
import pyb
from pyb import Timer
from pyb import Pin

# This tests the function of Pin.OUT_OD and Pin.AF_OD in pyb.Pin
# Connect a resistor of between 1k and 10k to pyboard pin X3 and 5V supply.
# In push pull_mode a '1' will read 3.3V, in open_drain mode it should read 5V.
print("pyb.Pin")
print("alternate function open_drain test, PWM 100%, pin X3 should be 5V")
# follwing test fails. Pin voltage is 3.3V
red_ext = Pin('X3', Pin.AF_OD, af = 2)
tim5 = Timer(5, freq=0.25)
red_ext_PWM = tim5.channel(3, pyb.Timer.PWM, pin=red_ext)
red_ext_PWM.pulse_width_percent(100)
pyb.delay(10000)

print("gpio open_drain test, writing 1, pin X3 should be 5V")
# following test passes. Pin voltage is 5V
red_ext = Pin('X3', Pin.OUT_OD)
red_ext.value(1)
```

```
import pyb
import machine
from pyb import Timer
from machine import Pin

# This tests the function of Pin.OPEN_DRAIN and Pin.ALT_OPEN_DRAIN in machine.Pin
# Connect a resistor of between 1k and 10k to pyboard pin X3 and 5V supply.
# In push pull_mode a '1' will read 3.3V, in open_drain mode it should read 5V.
print("machine.Pin")
print("alternate function open_drain test, PWM 100%, pin X3 should be 5V")
# follwing test fails. Pin voltage is 3.3V
red_ext = Pin('X3', Pin.ALT_OPEN_DRAIN, alt = 2)
tim5 = Timer(5, freq=0.25)
red_ext_PWM = tim5.channel(3, pyb.Timer.PWM, pin=red_ext)
red_ext_PWM.pulse_width_percent(100)
pyb.delay(10000)

print("gpio open_drain test, writing 1, pin X3 should be 5V")
# following test passes. Pin voltage is 5V
red_ext = Pin('X3', Pin.OPEN_DRAIN)
red_ext(1)
```

In stmhal/timer.c there is no support for open drain PWM.

```
///   - `mode` can be one of:
///     - `Timer.PWM` - configure the timer in PWM mode (active high).
///     - `Timer.PWM_INVERTED` - configure the timer in PWM mode (active low).
```

In the official docs there is a waiver that machine pin   may not be supported by all ports but the pyb library is specific to the pyboard and lists Pin.AF_OD as a valid mode
```
Pin.AF_OD - configure the pin for alternate function, open-drain;
```

So there appears to be a discrepancy between the docs and actual implementation

dhylands, 2017-02-08:
What's happening is that when you pass a pin= argument to the channel function, it configures the pin as AF_PP.
```
>>> from machine import Pin
>>> from pyb import Timer
>>> red_ext = Pin('X3', Pin.ALT_OPEN_DRAIN, alt = 2)
>>> red_ext
Pin(Pin.cpu.A2, mode=Pin.ALT_OPEN_DRAIN, af=Pin.AF2_TIM5)
>>> tim5 = Timer(5, freq=0.25)
>>> red_ext_PWM = tim5.channel(3, pyb.Timer.PWM, pin=red_ext)
>>> red_ext
Pin(Pin.cpu.A2, mode=Pin.ALT, af=Pin.AF2_TIM5)
>>> 
```
The only real purpose of the pin= argument is to ensure that the pin gets initialized in the appropriate ALT mode, and since you've already done it, you could simply drop the pin= argument.
```
>>> from machine import Pin
>>> from pyb import Timer
>>> red_ext = Pin('X3', Pin.ALT_OPEN_DRAIN, alt = 2)
>>> red_ext
Pin(Pin.cpu.A2, mode=Pin.ALT_OPEN_DRAIN, af=Pin.AF2_TIM5)
>>> tim5 = Timer(5, freq=0.25)
>>> red_ext_PWM = tim5.channel(3, pyb.Timer.PWM)
>>> red_ext
Pin(Pin.cpu.A2, mode=Pin.ALT_OPEN_DRAIN, af=Pin.AF2_TIM5)
>>> 
```

This the relevant C code that is switching the pin mode:
https://github.com/micropython/micropython/blob/181f7d145002731e5baec0f2c43ac57818447f8b/stmhal/timer.c#L891-L909

chrismas9, 2017-02-08:
@dhylands Thanks Dave, that worked. I will keep this open until I review the docs and tutorials I followed to see whether they need clarification. Chris

dpgeorge, 2018-12-04:
I assume the docs are ok.  If not feel free to reopen with suggestions how to improve them.

closed 2018-12-04 by dpgeorge