micropython/micropython #2855: WiPy 1.0 mp v1.8.7 - SD OSError: 19

federicomm, 2017-02-08:
From telnet REPL I have :
>>>os.listdir('/sd')
Traceback ......
    File .....
OSError: 19
From micropython forum kfricke give me the following answer:
"The was a change regarding the used FAT library
and filesystem mount points. " Then suggested to 
file an issue to githib.
The same suggestion I have had from pycom forum.
To be able to read the SD card I have than reflashed the v1.8.6 firmware
Thank

kfricke, 2017-02-08:
I was referring to issue #2850

dpgeorge, 2017-02-08:
@federicomm please can you detail your set-up: what hardware you use (eg expansion board), and what code you use to mount the sd card.

federicomm, 2017-02-09:
Hardware:
WiPy 1.0 plus expansion board ( 1st version )
SD card on expansion board
Power supply to WiPy by a power bank

Samsung KZoom smartphone with the following app to interact with WiPy :
Mocha Telnet for REPL
AndFTP to flash micropython firmware.

All was running without problems with micropython firmware v1.8.6 that I have reinstalled on WiPy 1.0 after the failure to read the SD card with the v1.8.7

===== boot.py on /flash ======
# boot.py -- run on boot-up
# can run arbitrary Python, but best to keep it minimal
from machine import UART
from machine import SD
import os
uart = UART(0, 115200)
os.dupterm(uart)
sd = SD()
os.mount(sd, '/sd')
=======================================
The main.py file is empty.

I only experiment to learn using the telnet RPL and following your documentation examples.

Thank for your quick feedback and help

dpgeorge, 2017-02-09:
Thanks for the info, should be fixed by ec7dc7f8d796b5b67772d1f40863d13fb5e19be2.

closed 2017-02-09 by dpgeorge

federicomm, 2017-02-09:
In the attachments there is the result with the February 9 micropython
firmware .
Regards

Connect to 192.168.1.1:23                                                       
Connected                                                                       
MicroPython v1.8.7-205-g181f7d1 on 2017-02-09; WiPy with CC3200                 
Login as: Password:                                                             
Login succeeded!                                                                
Type "help()" for more information.                                             
                                                                                
>>> os.listdir('/sd')                                                           
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
OSError: 19                                                                     
>>> dir()                                                                       
['sd', 'os', 'SD', 'UART', '__name__', 'uart']

federicomm, 2017-02-09:
SD reading after to have reinstalled on WiPy1 the v1.8.6 firmware

To complete my previous email
Regards and excuse me for the troubles.

federicomm, 2017-02-11:
Only to thank you and Dr.Damien George for the solution of the SD issue -
with firmware MicroPython v1.8.7-213- gaac2db9 on 2017-02-11;WiPy with
CC3200

Regards
Federico MONALDI