micropython/micropython #2857: compilation warnings

kamikaze, 2017-02-09:
Not sure if this is a bug... but today I've started to get this when compiling firmware:
```
arm-none-eabi-ld: warning: build-PYBV11/lib/libc/string0.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/math.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/thumb_vfp_sqrtf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/asinfacosf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/atanf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/atan2f.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/fmodf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/roundf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/log1pf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/acoshf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/asinhf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/atanhf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/kf_rem_pio2.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/kf_sin.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/kf_cos.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/kf_tan.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/ef_rem_pio2.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/erf_lgamma.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/sf_sin.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/sf_cos.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/sf_tan.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/sf_frexp.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/sf_modf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/sf_ldexp.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/sf_erf.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/wf_lgamma.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/lib/libm/wf_tgamma.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/system_stm32.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/flash.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_adc.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_adc_ex.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_can.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_cortex.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_dac.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_dac_ex.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_dma.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_flash.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_flash_ex.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_gpio.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_i2c.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_pcd.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_pcd_ex.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_pwr.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_pwr_ex.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_rcc.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_rcc_ex.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_rng.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_rtc.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
[... cut ...]
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_hal_uart.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_ll_sdmmc.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: build-PYBV11/hal/f4/src/stm32f4xx_ll_usb.o uses 32-bit enums yet the output is to use variable-size enums; use of enum values across objects may fail
```

kamikaze, 2017-02-09:
gcc-arm-none-eabi-5_4-2016q3

dpgeorge, 2017-02-09:
Please run `make clean` to remove old object files.  Then it should build ok.

closed 2017-02-09 by dpgeorge