micropython/micropython #2860: cc3200: WiPy1 ftp server not working

niallp, 2017-02-10:
I'm unable to build firmware for my wipy1 that has a working ftp server. Builds complete with no errors and the firmware seems to work fine otherwise (telnet server runs, a simple MQTT logger works) but the ftp server locks up just before prompting for the password.

Binaries from the nightly builds don't have this issue so evidently something wrong on my end. I've tried Ubuntu 16.04 using the gcc-arm-embedded from the PPA as well as a Debian Jessie system with the toolchain installed directly from the GNU ARM Embedded site, same problems with both on a variety of different versions from micropython/micropython.

niallp, 2017-02-15:
Just tried it on one of my Launchpad boards and it is acting the same ... telnet is fine, ftp locks up when logging in.

dmartauz, 2017-02-15:
Did you set your FTP client to passive mode with only one simultaneous connection and without encryption (plain FTP) ?

niallp, 2017-02-15:
Yes, only difference was local built firmware vs. those built elsewhere. 
Further digging and it appears to be the version of gcc used, 4.8.4 
works, 6.2.1 doesn't ... presently trying to narrow down which is last 
gcc to work. Might be 64 bit vs. 32 bit compilers, the latest gcc 
versions are native 64 bit ...

niallp, 2017-02-15:
latest working compiler: [https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2](url)
first broken build: [https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2](url)

the GCC v6 series all appear to be broken as well.

The version options are identical between working or not, unclear whether the issue is a bug in the toolchain itself or a latent bug in micropython triggered by the newer toolchains.

agatti, 2026-05-28:
Thanks, @robert-hh!

I'll close this then.