micropython/micropython #2864: esp8266 0x4016 works only with version 1.8.5

miticojo, 2017-02-12:
Hello,
I've tried to flash latest two stable and daily version without success on a bunch of esp8266 dev id 0x4016 but unfortunately without success. Once completed I got everytime reset with following message:
``` 
MicroPython v1.8.7-213-gaac2db9 on 2017-02-12; ESP module with ESP8266
Type "help()" for more information.
>>> prinFatal exception 4(Level1InterruptCause):
epc1=0x400005cb, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32120, room 16 
tail 8
chksum 0xdb
load 0x3ffe8000, len 1088, room 0 
tail 0
chksum 0xfc
load 0x3ffe8440, len 3000, room 8 
tail 0
chksum 0x18
csum 0x18
l��r��yr�'b����8�p�lp�������p�lll��r��  yr�'b��lb��p��p�������p��#l`��������rn�������l��b���b�����l�b��bl`ľ��8�ǒ���?b����lr����8����r����8���ffe8440, len 3000, room 8 
tail 0
chksum 0x18
csum 0x18
```

Have you any idea or suggestion to make it works?
I'm available to debug that error more deeply if needed to fix for all.

dpgeorge, 2017-02-13:
Please try to completely erase your flash before uploading new firmware (eg esptool.py erase_flash, or just "make erase").

miticojo, 2017-02-13:
Sure, I erased it before flash it.

dpgeorge, 2017-02-13:
My 0x4016 module works ok.  Did you use the most recent version of esptool.py?  And then also the "detect" parameter passed to the flash_size command-line arg?

miticojo, 2017-02-14:
ok, seems to be a problem of my modules. Thanks for your kind support!

closed 2017-02-14 by miticojo