micropython/micropython #2871: Add github tags for micropython/* projects

pfalcon, 2017-02-14:
Github recently introduced "tags" for projects. It would be nice to add them to projects of micropython organization. Example of tags is https://github.com/pfalcon/picoweb , a line below project description. @dpgeorge , if you go to https://github.com/micropython/micropython or similar, you'll see "Manage topics" link. Please add at least following topics:

* for micropython/micropython - "micropython", "python", "embedded".
* micropython/micropython-lib - "micropython"

It should not take more than 30s per project. Thanks. My concern is that list https://github.com/search?q=topic%3Amicropython&type=Repositories started with the right projects.

kamikaze, 2017-02-14:
noooo, my repo is the last one :( :D

kfricke, 2017-02-14:
How about a snychronized tag for contributed device "drivers"?

dpgeorge, 2017-02-15:
Ok, topics are now added.

> How about a snychronized tag for contributed device "drivers"?

Using "micropython" topic should be enough, but feel free to use "micropython-driver" if you like.

closed 2017-04-05 by pfalcon