micropython/micropython #2873: RFC: Consistent naming of long int/bignum tests

pfalcon, 2017-02-15:
We now have:

* int_big_*.py
* *_long.py
* *_bignum.py
* *_bigint.py

Taking int_big* as a template, I propose to rename the rest to have *_bigint.py suffix, they can be effectively skipped.

dpgeorge, 2017-02-16:
I think `basics/int_long.py` was written specifically to test the long-long implementation.  So it should have a different name so it can be included as tests on ports with long-long ints.

`basics/int_bytes_long.py` and `basics/slice_bignum.py` can certainly be renamed to have a bigint suffix.

I see no tests of the form `*bigint.py`.

pfalcon, 2017-02-16:
> I think basics/int_long.py was written specifically to test the long-long implementation. So it should have a different name so it can be included as tests on ports with long-long ints.

My initial idea was to separate small int vs all the rest, but if you think we should categorize 3 groups right away, can do it.

> I see no tests of the form *bigint.py.

Well, there would be, per the proposal. Or other alternative ideas? What about:

* int_big_*.py stays as is (at least for now, need to be reviewed re: long long vs MPZ compat)
* *_intlong.py for long-long friendly
* *_intbig.py for MPZ

dpgeorge, 2017-02-17:
> int_big_*.py stays as is

This is what I thought the initial proposal was ("... I propose to rename the rest...").  I'd be inclined to keep all those `int_big*` files named as they are (ie not rename them in anyway) and adjust others to suit that pattern.  Eg `int_long_*.py` are for int tests specific to long-long (so int_long.py is renamed to int_long_basic.py), and then `*_intlong.py` and `*_intbig.py` for tests that aren't int specific (eg the slice test).

pfalcon, 2017-03-03:
First step in this direction done: 89e570a5b4025976da66123032c58611f0f6cb17

pfalcon, 2017-03-07:
With https://github.com/micropython/micropython/commit/bdd48e67ee54163f195628ba6de476ca7984d327, there're no failing tests when running against unix port built with MICROPY_LONGINT_IMPL_NONE. MICROPY_LONGINT_IMPL_LONGLONG case was not tested and is subject of future work.

closed 2017-03-07 by pfalcon