micropython/micropython #2874: TCS34725 (colour sensor) library not working on Pyboard V1.1

smssmssms, 2017-02-15:
I've not been able to get the Adafruit TCS34725 library [(https://github.com/adafruit/micropython-adafruit-tcs34725/blob/master/tcs34725.py)](url) working on Pyboard V1.1, but it works fine on a V1.0 board; both with latest firmware downloaded 14/2/17.

On the V1.1 board, the sensors seem to be active and visible on the I2C bus, but when you try to read them`sensor.read(1)` you get Errno 5 as though there is no sensor there. 

Bit more background info here:
[https://forums.adafruit.com/viewtopic.php?f=60&t=111774](url)
[http://forum.micropython.org/viewtopic.php?f=14&t=3019](url)

peterhinch, 2017-02-16:
Both those URL's are 404's. Try
[adafruit forum](https://forums.adafruit.com/viewtopic.php?f=60&t=111774&p=559246&hilit=TCS34725#p559246)
[uPython forum](http://forum.micropython.org/viewtopic.php?f=14&t=3019&p=17940&hilit=TCS34725#p17940)
I'd be surprised if there is a difference between these Pyboard variants. I'd look at the wiring first. Make sure the wires are short. It might be worth trying 1K pullups (to 3.3V) on the I2C lines. The Pyboards have 4K7 but stronger pullups can yield faster rise times.

dpgeorge, 2017-02-17:
> I'd be surprised if there is a difference between these Pyboard variants.

Yes, there is hardly any difference between PYBv1.0 and PYBv1.1.  The exact differences are (going from v1.0 to v1.1): high-speed crystal changed from 8MHz to 12MHz; LDO changed from MCP1802 to MCP1703; USB VBUS power diode replaced with silicon diode; JST battery connector pads added with protection FET.

Maybe try some decoupling capacitors on the power rails of the sensor.

pfalcon, 2017-05-14:
Assuming above suggestions clarified it. Feel free to reopen with additional info.

closed 2017-05-14 by pfalcon