micropython/micropython #2879: lcd160pr.py touch detection fails if called in a tight loop

peterhinch, 2017-02-17:
To replicate paste the following, then touch the screen for a long period:
```python
import lcd160cr
lcd = lcd160cr.LCD160CR('X')
lcd.set_orient(lcd160cr.PORTRAIT)
touched = False
while True:
    nt = lcd.get_touch()
    if not nt[0] == touched:
        print('touch:', nt)
    touched = nt[0]
```
Background: I'm porting my SSD1963 GUI to the lcd160cr. Touch detection is performed in a uasyncio coroutine which runs with a minimal delay ``await asyncio.sleep_ms(0)``. It fails periodically.

dpgeorge, 2017-02-17:
Thanks for the report.  I actually came across the same bug earlier today but from a different use-case (tight loop using lcd.is_touched() and sending out a big frame buffer over SPI).

peterhinch, 2017-02-18:
I've looked at this some more: the code looks fine so I wonder if it's a hardware issue?

Experimentation suggests it's related to time and to the number of successive ``get_touch()`` calls you make. Requiring five successive identical results with a 10ms wait between each call seems reliable. With no delay in a tight loop I needed ~50 identical results for reliable operation. This gives good responsiveness and reliable operation on my hardware:
```python
    async def _touchtest(self):
        touch_panel = Screen.tft
        was_touched = False
        while True:
            await asyncio.sleep_ms(0)
            touched, x, y = touch_panel.get_touch()
            if touched != was_touched:
                for _ in range(5):
                    await asyncio.sleep_ms(10)
                    touched, x, y = touch_panel.get_touch()
                    if touched == was_touched:
                        break
                else:
                    was_touched = touched
                    # handle touch and untouch events here
```
One solution might be for ``get_touch()`` to have a ``block`` parameter defaulting True. By default it blocks until a correct result is acquired. Otherwise the user - perhaps using uasyncio - takes responsibility.

peterhinch, 2017-02-19:
Further testing showed the above solution is a kludge. Simplifying the code produced a bad outcome for reasons which elude me. Is there a hardware problem (decoupling?).

Guidance sought!

chuckbook, 2017-02-19:
@peterhinch Is there a way to get an example of your GUI using the lcd160cr?

peterhinch, 2017-02-21:
A fix now exists in the form of a firmware update to the display. Official release pending.

closed 2017-02-21 by peterhinch