micropython/micropython #2881: lcd160cr.py get_pixel() and set_pixel() methods are not mutually compatible

peterhinch, 2017-02-17:
The following produces a surprising (to me, at least) colour change:
```python
import lcd160cr
lcd = lcd160cr.LCD160CR('X')
lcd.set_orient(lcd160cr.PORTRAIT)
lcd.set_pos(0, 0)
lcd.set_text_color(lcd.rgb(255, 0, 0), lcd.rgb(0, 0, 0))
lcd.set_font(1)
lcd.write('Hello MicroPython!')
print('touch:', lcd.get_touch())
for y in range(10):
    for x in range(60):
        lcd.set_pixel(x, y, lcd.get_pixel(x, y))
```

chuckbook, 2017-02-17:
It looks like there is a bug in the get_pixel method of lcd160cr.py. Line 252 should read:

`return self.buf[3][1] + (self.buf[3][2] << 8) # missing parenthesis around shift operator`

peterhinch, 2017-02-17:
Yes, that fixes the issue with my application.

dpgeorge, 2017-02-21:
Now fixed, see link above.

closed 2017-02-21 by dpgeorge