micropython/micropython #2882: lcd160cr buggy repl output

kamikaze, 2017-02-17:
When setting repl otput to LCD160CR and printing in infinite loop current time in micros with different font sizes following occurs:
[video1](https://www.youtube.com/watch?v=EHpVY45NN_U)
[video2](https://www.youtube.com/watch?v=mM7DaQurGBI)

Notice that periodically first letter of largest font is being partially erased

chuckbook, 2017-02-17:
The scroll is triggered once a (printable) character is processed. So after a linefeed from a small font the vertical space may not be sufficient for the new big font. You may observe this by using smooth scroll.
`lcd.set_font(i & 3, scroll=1)`

kamikaze, 2017-02-17:
but it does the same sequence of font sizes and data length in a loop. but first char fails randomly

chuckbook, 2017-02-17:
It isn't randomly, it happens every 4th appearance of a big C.

dpgeorge, 2017-02-21:
@kamikaze can you please provide a minimal script which has the problem, for both first and second video?

chuckbook, 2017-02-21:
This exhibits the symptom:

```
import lcd160cr_a as lcd160cr

lcd = lcd160cr.LCD160CR("YX")             # replace YX with actual socket
lcd.set_orient(lcd160cr.LANDSCAPE)
i = 0
lcd.set_text_color(0x7e0, 0)
while True:
    lcd.set_font(i & 3, scroll=1)
    i += 1
    lcd.write("Current time:\r\n%d\r\n" % (pyb.micros()))
    pyb.delay(1000)

```

jonnor, 2024-09-29:
Is this issue still present in recent MicroPython version?

Note that the driver is now in micropython-lib, and likely needs to be installed separately. https://github.com/micropython/micropython-lib/tree/master/micropython/drivers/display/lcd160cr

peterhinch, 2024-09-29:
The LCD160CR has been unavailable for a very long time. I think the plan was to replace with a new design.

closed 2025-03-03 by jonnor