micropython/micropython #2884: UDP Asynccall back not functioning

AgentSmithers, 2017-02-18:
UDP callback not registering within the Python call

s.setsockopt(socket.SOL_SOCKET, _SO_REGISTER_HANDLER, callback_function) # register callback

Does exec_user_callback(socket); need to be added to the function below?

STATIC void _lwip_udp_incoming(void *arg, struct udp_pcb *upcb, struct pbuf *p, ip_addr_t *addr, u16_t port) {
     lwip_socket_obj_t *socket = (lwip_socket_obj_t*)arg;
 
     if (socket->incoming.pbuf != NULL) {
         // That's why they call it "unreliable". No room in the inn, drop the packet.
         pbuf_free(p);
     } else {
         socket->incoming.pbuf = p;
         socket->peer_port = (mp_uint_t)port;
         memcpy(&socket->peer, addr, sizeof(socket->peer));
     }
 }

dpgeorge, 2017-02-20:
This callback mechanism was added to get WebREPL working and is not really intended for general use, so that's why the UDP part is not implemented.

@pfalcon I don't recall what the long-term plan for socket callbacks was, do you?  If we want to make this a standard feature on all ports then it requires some work, because in general a port would only provide high-level socket interface, not low level like we have in esp8266.

pfalcon, 2017-02-20:
> This callback mechanism was added to get WebREPL working and is not really intended for general use, so that's why the UDP part is not implemented.

That indeed was the one and only plan for it. It is undocumented, internal feature, users are discouraged to use it for anything but an intended purpose (dupterm optimization for webrepl), and then there's no guarantee that it's not changed in the future.

robert-hh, 2017-02-20:
It is useful and used, so please keep that mechanism.

AgentSmithers, 2017-02-21:
Everyone, I added the line of code  and UDP is now functioning correctly with the Callback function.

robert-hh, 2017-02-21:
Please sent a pull request.

AgentSmithers, 2017-02-22:
I did, I hope I did it correctly. Thank you everyone.

pfalcon, 2017-05-06:
Hopefully, https://github.com/micropython/micropython/issues/2884#issuecomment-281031884 explains the situation with this. Closing.

closed 2017-05-06 by pfalcon