micropython/micropython #2886: Crash when trying to insert tuple into list

herm, 2017-02-19:
Test case:
```
l = [1,2,3,4]
l[0:1] = (1, 2, 3)
```
Reboots the system with MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09; ESP module with ESP8266

Trying to do the same thing with a list works just fine:
```
l = [1,2,3,4]
l[0:1] = [1, 2, 3]
```

I understand that this might not be supported in micropython but it should then print an error message instead of just rebooting.

stinos, 2017-02-19:
Good catch, segfaults in windows build as well.

dpgeorge, 2017-02-20:
Thanks for the report, definitely a bug :)  The issue was that the code assumed the RHS was a list.  It's now fixed (see above link) and you can now have both list and tuple on the RHS of a list-slice-assignment.

Note on code size increase: the fix uses the existing helper function to check and extract the values from a list/tuple, and costs 12 bytes of code on stmhal and esp8266.  Not using the helper and just checking for a list costs 24 bytes.

closed 2017-02-20 by dpgeorge