micropython/micropython #2889: Document need (or no need) to save & restore registers in asm_xtensa functions

ARF1, 2017-02-20:
I am discovering micropython for the esp8266 and am absolutely astonished at what is possible already!

One thing I am struggling with is the `machine.asm_xtensa` decorator which so far is undocumented.

Most things can be understood by analogy to the documented `machine.asm_thumb` decorator, but one thing absolutely stumps me:

Does one have to save and restore any registers one modifies in an `asm_xtensa` function? If so, are there any convenient solutions for doing so, or does one have to all the stack pointer algebra oneself? The `asm_thumb` instruction `push({...})` looks great!

dpgeorge, 2017-02-21:
> One thing I am struggling with is the machine.asm_xtensa decorator which so far is undocumented.

Yes, this is an advanced feature that's stil WIP and hence undocumented.

> Does one have to save and restore any registers one modifies in an asm_xtensa function?

No.  The registers a0,a12,a13,a14 are automatically saved and restored for you.

Note that a0 is the return address, a1 is the stack pointer, a2-a5 are the arguments, and a2 is the return value.

ARF1, 2017-02-21:
Thanks for the clarification. I saw the a0, a12, a13, a14 saving operation in the code but I was not sure, whether somewhere else, more registers were get saved.

In my opinion the asm-decorator is an awesome feature! It is what convinced me that python on a microcontroller is a good idea. - You should really promote this feature heavily in my opinion. It is a game-changer!

ARF1, 2017-02-22:
@dpgeorge 

> No. The registers a0,a12,a13,a14 are automatically saved and restored for you.

And all other registers are caller-saved as per xtensa CALL0 ABI, right?

If so, I have a follow-up question: why is `a15` not saved? The Xtensa ISA reference manual specifies in section 8.1.2:

> The stack pointer in register a1 and registers a12–a15 are callee-saved

If you are are taking care saving/restoring the callee-saved registers to make the learning curve shallower for new users, shouldn't `a15` be saved too? What am I overlooking?

dpgeorge, 2017-02-28:
> If you are are taking care of saving/restoring the callee-saved registers to make the learning curve shallower for new users, shouldn't a15 be saved too? What am I overlooking?

The assembler backends were originally designed to support the native emitters (@micropython.native and @micropython.viper).  These emitters require a few callee-save registers, and in fact 3 are enough.  The entry/exit code therefore only saves 3 registers, and the same code is used for entry/exit for the inline assembler.  Also, since the stack needs to be 16-byte aligned, saving only 4 (instead of 5) makes it a bit more efficient in RAM usage.

So that's the reason why a15 is not saved.