micropython/micropython #2890: [esp8266] OSError: [Errno 113] EHOSTUNREACH only when executing connect() in main.py

TheMeaningfulEngineer, 2017-02-20:
Hi

For the following code, 

```
import machine
import network
from umqtt.simple import MQTTClient

led = machine.Pin(D1, machine.Pin.OUT)
button = machine.Pin(D0, machine.Pin.IN)

sta_if = network.WLAN(network.STA_IF)
ap_if = network.WLAN(network.AP_IF)

ap_if.active(False)
sta_if.active(True)
sta_if.connect('my-Lan', 'My-passwoed')

client = MQTTClient("esp-button", "192.168.1.199")
client.connect()

while True:
    if button.value() == 1:
        led.value(1)

    else:
        led.value(0)
```

I'm experiencing the error below:

```
Traceback (most recent call last):
  File "main.py", line 17, in <module>
  File "umqtt/simple.py", line 56, in connect
OSError: [Errno 113] EHOSTUNREACH

```
Line 17 is `client.connect()` and [line 56](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py#L56) is `self.sock.connect(self.addr)`.

The peculiarity is that the exact same code can be run without errors when imputed step by step into the interactive shell. Actually, as soon as the exception occurs and the control returns to the shell I can run `client.connect()` without errors (returns 0) which is shown in the MQTT broker log.

The version used is:  `MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09` on ESP8266

TheMeaningfulEngineer, 2017-02-20:
When I set it in a loop:

```
import machine
import network
from umqtt.simple import MQTTClient

led = machine.Pin(D1, machine.Pin.OUT)
button = machine.Pin(D0, machine.Pin.IN)

sta_if = network.WLAN(network.STA_IF)
ap_if = network.WLAN(network.AP_IF)

ap_if.active(False)
sta_if.active(True)
sta_if.connect('o2-WLAN42', '94796AV9436KQQN3')

client = MQTTClient("esp-button", "192.168.1.199")

while True:
    try:
        client.connect()
    except Exception:
        continue
    break

while True:
    if button.value() == 1:
        led.value(1)

    else:
        led.value(0)
```

there seems to be an additional pattern showing up

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)^M
^M
load 0x40100000, len 32028, room 16 ^M
tail 12^M
chksum 0x40^M
ho 0 tail 12 room 4^M
load 0x3ffe8000, len 1092, room 12 ^M
tail 8^M
chksum 0x17^M
load 0x3ffe8450, len 3000, room 0 ^M
tail 8^M
chksum 0x56^M
csum 0x56^M

```

I could only notice that by logging the serial output as it constantly restarted:
`picocom -b 115200 /dev/ttyUSB0 main.py | tee -a log `

dpgeorge, 2017-02-22:
Regarding the first error: the issue is that sta_if.connect() returns before the connection to the WLAN is made.  So you are trying to use the wifi without it first being set up.  To fix this put a `while not sta_if.isconnected(): pass` loop ofter the call to connect().

Regarding the second error (with the loop and crash): it's probably because the code is trying to make lots of sockets which are not being closed, and then the wifi connects and there is no memory left.  The fix for the first issue should solve this as well.

TheMeaningfulEngineer, 2017-02-23:
That was it, thanks :)

closed 2017-02-23 by TheMeaningfulEngineer

KhawajaMoeez, 2018-10-20:
Can anyone of you please help me regarding this error? When I try to run Python script of getting sensor data from dht-11 after enabling Wi-Fi on esp32, I'm getting Oserror errno113 EHOSTUNREACH.

Please assist me asap.

peterhinch, 2018-10-21:
Support queries are best raised in [the forum](https://forum.micropython.org/) where they will be seen by more people. I doubt anyone will be able to help without seeing your code.

KhawajaMoeez, 2018-10-21:
from time import sleep
from umqtt.simple import MQTTClient
from machine import Pin
from dht import DHT11

SERVER = '192.168.137.152'  # MQTT Server Address
CLIENT_ID = 'ESP32_DHT11_Sensor'
TOPIC = b'temp_humidity'

client = MQTTClient(CLIENT_ID, SERVER)
client.connect()   # Connect to MQTT broker

sensor = DHT22(Pin(15, Pin.IN, Pin.PULL_UP))

while True:
    try:
        sensor.measure()   # Poll sensor
        t = sensor.temperature()
        h = sensor.humidity()
        if isinstance(t, float) and isinstance(h, float):  # Confirm sensor results are numeric
            msg = (b'{0:3.1f},{1:3.1f}'.format(t, h))
            client.publish(TOPIC, msg)  # Publish sensor data to MQTT topic
            print(msg)
        else:
            print('Invalid sensor readings.')
    except OSError:
        print('Failed to read sensor.')
    sleep(4)

peterhinch, 2018-10-21:
I assume the error is provoked by this line:
```python
client.connect() # Connect to MQTT broker
```
If so, I would first try pinging the IP address, and connecting to the broker using a PC client. If this succeeds you can be reasonably confident of the broker.

If the broker is OK it's likely that your ESP8266's WiFi connection hasn't been established soon enough. See the earlier messages in this thread for solutions.

This is the MicroPython bug tracker and there is no evidence so far that you're encountering a MicroPython bug. As I said before, it's in your interests as well as those of the MicroPython devs that support queries are raised in [the forum](https://forum.micropython.org/).

KhawajaMoeez, 2018-10-22:
Thank you so much for the help Peter. I was messing with the IP address. Somehow LAN interface IP was passed.