micropython/micropython #2892: stmhal Frozen bytecode: no longer works with lib subdirectory

peterhinch, 2017-02-21:
I have stmhal/modules/lib containing the uasyncio library. I have been building successfully with
```
make -j 8 BOARD=PYBV11 FROZEN_MPY_DIR=modules
```
Today, though it compiled the files
```
...
MPY modules/lib/uasyncio/test2.py
MPY modules/lib/uasyncio/core.py
MPY modules/lib/uasyncio/__init__.py
MPY modules/lib/uasyncio/test.py
MPY modules/lib/uasyncio/queues.py
MPY modules/lib/collections/deque.py
...
```
I can no longer import uasyncio at the REPL. Modules in stmhal/modules (i.e. not in the subdirectory) are OK.

dpgeorge, 2017-02-22:
By putting the files in the "lib/" subdir you have implicitly created a package called "lib".  So you need to do "from lib import collections", etc.  Alternatively, add to your path: `sys.path.append('lib')`.

I wonder though, why did this work before?  What version of uPy did it work with?

peterhinch, 2017-02-22:
It still works. With an SD card having a ``lib`` subdirectory copied from ``~/.micropython`` and containing uasyncio:
```
MicroPython v1.8.1-1410-g71ae3f3-dirty on 2017-02-22; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
>>> 
PYB: sync filesystems
PYB: soft reboot
MicroPython v1.8.1-1410-g71ae3f3-dirty on 2017-02-22; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import uasyncio
>>> 
```
I can also issue ``from lib import uasyncio`` and agree that this is the correct syntax.

I evidently mis-diagnosed the issue with freezing uasyncio. Apologies. I think there is something in the way uasyncio is configured which is causing the problem. With a ``lib`` directory under my frozen modules directory, and no copy of uasyncio in the normal filesystem, I get the following:
```
MicroPython v1.8.1-1410-g71ae3f3-dirty on 2017-02-22; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
>>> from lib import uasyncio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lib/uasyncio/__init__.py", line 4, in <module>
ImportError: no module named 'uasyncio'
>>> 
```
Evidently it's finding ``__init__.py`` so the build proceeded OK.

peterhinch, 2017-02-27:
I've now revisited this. When you install ``uasyncio`` to the Unix build with ``upip`` it creates a directory ``~/.micropython/lib`` and installs ``uasyncio`` as files and subdirectories there. To successfully freeze ``uasyncio`` it is necessary to copy the files and subdirectories to the ``FROZEN_MPY_DIR``. It can then be imported with
```python
import uasyncio
```
(As per the Unix build).
My mistake was to copy ``lib`` there, as copying ``lib`` to an SD card had worked. It would be good if the documentation detailed the way to install a library to target hardware, both normally and as frozen bytecode. I'd be happy to do this but I'd appreciate guidance as to where in the docs this information belongs.

closed 2017-02-27 by peterhinch