micropython/micropython #2893: Add push/pop asm_xtensa convenience functions

ARF1, 2017-02-21:
It would be nice if `asm_xtensa` were to provide push/pop convenience functions for saving/restoring registers.

In `asm_thumb` one can write very clean code (ignoring the wrong register names for illustration purposes):
```
@micropyhton.asm_thumb():
def foo():
    push({a5, a6, a7, a8, a9, a10, a11})
    # do your stuff
    push({a5, a6, a7, a8, a9, a10, a11})
```

The same with `asm_xtensa` is a nightmare to write and easy to mess up:
```
@micropyhton.asm_thumb():
def foo():
    # calculate stack pointer to make space for registers
    # but: must be in multiples of 16!
    addi(a1, a1, -32)
    # store registers
    s32i(a5, a1, 0)
    s32i(a6, a1, 4)
    s32i(a7, a1, 8)
    s32i(a8, a1, 12)
    s32i(a9, a1, 16)
    s32i(a10, a1, 20)
    s32i(a11, a1, 24)

    # do your stuff

    # restore registers
    l32i(a5, a1, 0)
    l32i(a6, a1, 4)
    l32i(a7, a1, 8)
    l32i(a8, a1, 12)
    l32i(a9, a1, 16)
    l32i(a10, a1, 20)
    l32i(a11, a1, 24)
    # move stack pointer back
    addi(a1, a1, 32)
```
And here I am not even checking for stack underflow/overflow!

dpgeorge, 2017-02-22:
The thing with the asm_thumb instructions push/pop is that they are *single* instructions even when you have multiple arguments (the registers form a bitmap of those to push/pop).  So they are not convenience opcodes but real ones.

I'd rather not add convenience opcodes because then they need to be documented, and take up code space.

ARF1, 2017-02-22:
@dpgeorge 
> I'd rather not add convenience opcodes because then they need to be documented, and take up code space.

Just to make sure I understand: you are worried that the "convenience opcodes" would not increase the size of the emitted code but the code of the compiler itself, right?

If so, is there any way of installing a sort of "hook", so that user's can define their own "convenience opcodes" instead? I.e. a sort of a user-defined pre-compiler? Then this code size "cost" could be at the discretion of the user.

Maybe with an interface similar to [ast.NodeTransformer](https://docs.python.org/3/library/ast.html#ast.NodeTransformer) where for every ast node, a separate method (with method name = ast node name) gets called and the node replaced by the return value.

This would allow creating some awesome assembler macros:

```
class my_xtensa_prec(object):
    free_words_stack = 0

    def push(registers):
        ret = []
        if self.free_words_stack < 4 * len(registers):
            required_words_stack = len(registers) - self.free_words_stack
            # 16-byte align
            required_words_stack = ceil(required_words_stack / 4) * 4
            ret.add('addi(a1, a1, -%d)' % required_words_stack * 4)
            self.free_words_stack += required_words_stack
        stack_offset = (self.free_words_stack - len(registers)) * 4
        for n, reg in enumerate(registers):
            ret.add('s32(%s, a1, %d)' % (reg, stack_offset+n*4))
        return ret

@micropython.asm_xtensa(precompiler=my_xtensa_prec):
def some_fn():
    push({a2, a3, a4})
``` 

My naive idea (without having really understood the micropython compiler) would be:

1. the compiler checks whether a `precompiler` argument was supplied to the decorator
2. if yes, do a pre-compilation pass during which
4. for each `opcode(*args)`, it would call `precompiler.opcode(*args)`
3. and 'replace the `opcode(*args) ' by the returned opcodes

I am not sure whether something like this is feasible without huge effort. Of course this would add some code space as well, but at least in return we would get a very versatile feature.

dhylands, 2017-02-22:
Another option would be to write a pre-processor that runs on the host, which takes your python source code and expands it.

jonnor, 2024-07-02:
Is this still a desirable feature?

closed 2026-05-25 by dpgeorge