micropython/micropython #2896: Error when importing .mpy module

honzaskypala, 2017-02-22:
I compiled an mpy module and tried to import it on an ESP8266 board. Unfortunately I got an error

```
>>> import testmpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: incompatible .mpy file

```

First I thought it was probably an issue of the ESP8266 board, but then I cloned the whole micropython git source (v1.8.7-306-gb0a6dda) on my debian machine, successfully compiled it (both micropython and mpy-cross) and I getting the same error there. Then I made just a very simple module, compiled it using mpy-cross, but still getting the same error on import.

The testing module:

```
def susum(a, b):
    return a + b
```

I searched for the same issue, found [#2540](https://github.com/micropython/micropython/issues/2540), but this was about incompatible versions of micropython and mpy-cross. This should not be the case here, as I'm using both tools from just one git clone.

Any help?

robert-hh, 2017-02-22:
The .mpy file format was recently changed. So it is important to use mpy-cross firmware from the same version, like you seemed to have done. I recompiled a firmware with embedded frozen bytecode after that transition, and it worked. But I did not try separately compied .mpy files from the file system. Are you sure that you compiled your files with the recent build and that you copied the right file to the esp8266? I'm asking because these kind of glitches happen to me quite often.  I'll try myself this evening.

honzaskypala, 2017-02-22:
I wanted to first solve it on my debian machine, as it does not work even there. I just git-cloned the repository yesterday, and built both micropython and mpy-cross from it, without any issues. Micropython runs on my debian machine just fine, I can work in REPL. If I import the module as *.py file, it works great.

When I compile the syme .py file using the mpy-cross, then I get the *.mpy file. But when I try to import it to micropython, still on the same debian machine, it produces the error mentioned above.

I attach the compiled *.mpy file.

[testmpy.zip](https://github.com/micropython/micropython/files/793010/testmpy.zip)

Regarding my ESP8266 board, it reports _MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09; ESP module with ESP8266_
Version compiled on my debian machine is _MicroPython v1.8.7-306-gb0a6dda on 2017-02-21; linux version_

robert-hh, 2017-02-22:
Confirmed for the Linux and Windows versions of micropython. I'll check later on the ESP8266. But your ESP8266 definitely needs an update to get compatible. It also happens for V1.8.7.242, which is dated one day before the .mpy file change was flagged.

peterhinch, 2017-02-22:
Cross compiling is not particularly well documented. I've been puzzling over it in the context of cross compiling code which uses the native emitter. There is surely a mechanism to tell mpy-cross the target architecture - it must be possible as you can freeze such modules as bytecode.

The help file has ``-mcache-lookup-bc : cache map lookups in the bytecode``. I've no idea what that means, but it is necessary if the file is to run on the Unix build.

robert-hh, 2017-02-22:
I tested again the latest build with .mpy files from the file system, and it works. So it's more a documentation than a code issue. For PC builds (Windows & Unix), use the option 
`-mcache-lookup-bc. `
That matches the definition
`#define MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE (1)`
in mpconfigport.h.

honzaskypala, 2017-03-04:
Hi,

I confirm that when compiling with `-mcache-lookup-bc`, it works fine in Unix.

Also I confirm that when I flashed the latest MicroPython to ESP8266 _MicroPython v1.8.7-347-g3ab6aa3 on 2017-03-04; ESP module with ESP8266_, importing the mpy module on the board works fine as well.

Thank you all for your help.

closed 2017-03-04 by honzaskypala

spartacusxxxxxx, 2018-01-19:
What is the key for compilation you used for ESP 8266? Thanks for the answer