micropython/micropython #2897: lcd160cr get_line() method produces unexpected bytearray offset

peterhinch, 2017-02-22:
```python
import lcd160cr
buf = bytearray(20)
lcd = lcd160cr.LCD160CR('Y')
lcd.set_orient(lcd160cr.LANDSCAPE)
x = 5
y = 1
lcd.set_pixel(x, y, lcd.rgb(255,255,255))
lcd.set_pixel(x + 1, y, lcd.rgb(255,255,255))
lcd.get_line(x, y, buf)
print(buf)
```
Outcome
```
bytearray(b'\x0c\xff\xff\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
```
I tried various values of x and y, and both landscape and portrait modes: the result is always displaced by one byte. The contents of byte 0 seems to depend on orientation.

peterhinch, 2017-02-23:
Given a code sample from Thomas it seems this is by design. The behaviour is not intuitively obvious: I think the docs should clarify.

It would be interesting to know the purpose of this byte.

A typical use case for get_line() is to save a region for later restoration. Discarding the byte (to enable fast restoration via show_framebuf) adds buffer copying code implying a performance hit.

dpgeorge, 2017-05-17:
In commit f351c6db5e4c0bf75d5d647e69a047da739606a8 the get_line() method is "fixed", docs are updated to describe its behaviour, and screen_dump() is enhanced to allow to dump an arbitrary rectangular region.

closed 2017-05-17 by dpgeorge