micropython/micropython #2898: Allow creating const from 32-bit integer/hexadecimal

ARF1, 2017-02-22:
When writing code referencing memory one often specifies 32-bit memory addresses like `0x60000000`.

This would be a good case for using `const` to keep ram usage down but unfortunately it seems impossible to do so:

```
>>> a = const(0x8)                                                              
>>> a = const(0x60000e00)                                                       
Traceback (most recent call last):                                              
  File "<stdin>", line 2                                                        
SyntaxError: constant must be an integer                                        
```

The error message is slightly confusing as even explicitly specifying the value as an integer, does not help:
```
>>> a = const(int(0x60000e00))                                                  
Traceback (most recent call last):                                              
  File "<stdin>", line 2                                                        
SyntaxError: constant must be an integer                                        
```

dhylands, 2017-02-23:
I think what it really means is that the constant must be a small integer, which for platforms like stmhal means that it has to be a 31-bit signed number, which indirectly implies that when viewed as a 32-bit number the top 2 bits must be equal. 0x60000000 doesn't fit into a small int, which means that an mpz object needs to be allocated, which I think negates any advantage of declaring it const.

pfalcon, 2017-02-23:
Offtopic:

@ARF1 : Cute work on http://esp8266-re.foogod.com/w/index.php?title=Memory_Map&diff=prev&oldid=420 , etc. (assuming it's you). I hope you're aware of https://groups.google.com/forum/#!topic/esp8266-re/SJicbstg5TQ

dpgeorge, 2017-02-24:
This is now fixed in 74f4d2c65945d727821503e768c71f4e6b37f471.

closed 2017-02-24 by dpgeorge