micropython/micropython #2899: Connect to WiFi via BSSID (and not the SSID)

mitchins, 2017-02-22:
Forgive me if I'm wrong but this stumped me yesterday.

As per the documentation I read carefully connect method of the NIC driver only allows for SSID specification, however I had several access points of differing strength (which I could clearly see in the scan() method) but I could not provide the BSSID and it was connecting to the poorest one for some reason which was woeful.

Is there a way to connect with BSSID that is not explicitly documented, or is there a way we can implement it? I am not familiar with the HAL for the network chip, but it seems NodeMCU have added it at some point so it must be possible.

I suppose even if we could specify the channel that could allow us to do some shuffling, but BSSID would be ideal as it's returned in the scan along with signal strength.

dpgeorge, 2017-02-23:
There is a patch for specifying the BSSID in #2737.

mitchins, 2017-02-23:
Thanks, that is helpful to know, however it looks bleak around the contentious changes of connect()'s protocol.

Adam5Wu, 2017-03-16:
Agreed, the current wlan_connect API implementation need some refactoring and become more like that of cc3200's. Otherwise, future extension will always have this kind of problem.

But this is out of the scope of PR #2737, and I also have not enough know-how to make such change.

If someone can make the refactoring, I can always adopt or make new PR for this issue.

mitchins, 2017-08-27:
Just wondering if there's any news on this issue? If I have multiple base stations it would be nice to connect to the strongest one intelligently otherwise each AP needs a unique name which is less elegant if shared network with computers/mobiles.

pfalcon, 2017-08-27:
Yeah, I wanted to bring up this. Now we have the (first iteration of) generic networking API spec, and it says (implies) that connection by BSSID would use following syntax (http://docs.micropython.org/en/latest/pyboard/library/network.html#network.connect):

~~~
wlan.connect(None, "password", bssid=b"\x00\x01\x02\x03\x04\x05")
~~~

Is this good enough, or needs further elaboration (risking to complicate the "minimal" implementation of networking API)?

@dpgeorge ^

mitchins, 2017-08-27:
I think it’s great, functional and simple but scalable. I know this was the big hang up, and finally it’s solved. 

Is this actually implemented? What’s the version status of this protocol?

dpgeorge, 2017-10-05:
> Is this good enough, or needs further elaboration

That should be ok, but note that the SSID still needs to be specified when specifying the BSSID (at least on esp8266).

> Is this actually implemented?

See #3349.

dpgeorge, 2017-10-10:
Connecting via BSSID on esp8266 is now possible, see 6db132e1302dcca4671e0d67b872ca226c54682f

closed 2017-10-10 by dpgeorge

mitchins, 2017-10-10:
That’s awesome I’ll give it a go.
I’m working on a wifi settings manager that’ll abstract connecting to the highest strength network of same name

mitchins, 2018-07-07:
@dpgeorge I am curious how this affects ESP32.
Even if it is not supported, shouldn't the abstract NIC confirm to the same interface.
Just ran on 
MicroPython v1.9.4 on 2018-05-11; ESP32 module with ESP32
and got:
TypeError: function does not take keyword arguments
when calling this connect() with bssid

So even if it didn't do anything with it, shouldn't it gracefully accept kwargs?

dpgeorge, 2018-07-09:
> So even if it didn't do anything with it, shouldn't it gracefully accept kwargs?

I think it would lead to more troubles if it ignored an argument that it didn't support.  But, anyway, I added support for using `bssid` on esp32 in 9c8141f07e034983fe7aa37b3940afbd565730a3

mitchins, 2018-07-09:
@dpgeorge that's excellent. I'd started checking out the ESP-IDF and looking at what it would take in order to do it myself, seems like ESP-IDF is quite straight forward in this regard from the docs. 
Seems like you also added an extra check that the args are not None before memcpy, which even though len should've been 0, makes sense to be safer. 

Thanks, I'm quite happy this is in, ... I'll wait for next time to submit my own PR :)

dpgeorge, 2018-07-09:
> I'll wait for next time to submit my own PR

If you'd like to tackle a not-too-difficult problem, try issue #3913

jimmo, 2022-07-14:
> i feel necro for bringing this up here again, but whats about the bssid parameter now for connect()?

@WaGi-Coding no need to reply to multiple issues for the same thing.

See https://github.com/micropython/micropython/issues/7207 -- this is supported on many ports, the key is that you need to use positional arguments for ssid and key, then kwarg (or positional) for bssid. I will update the docs and examples.

Jibun-no-Kage, 2022-08-01:
Does this work for Pico W, i.e. rp2 support?  I can't seem to get it work all on a Pico W.