micropython/micropython #2900: ntptime.py: consider changing 'host' to 'NTP_HOST'

skimj, 2017-02-23:
Would you consider changing the 'host' variable to 'NTP_HOST' in the ntptime.py script? It would be consistent with the other NTP_ variables and would help avoid a very generic 'host' variable when set elsewhere to override the script default.

pfalcon, 2017-02-28:
> would help avoid a very generic 'host' variable set elsewhere to override the script default.

Please elaborate on that.

skimj, 2017-03-09:
I accidentally broke the ntptime script because I set a variable named 'host' for a completely different use without realizing it was used for ntptime. It's not critical, I'm simply suggesting that 'NTP_HOST' would be more descriptive and less likely to be used inadvertently.

pfalcon, 2017-03-18:
> I accidentally broke the ntptime script because I set a variable named 'host' for a completely different use without realizing it was used for ntptime.

Sorry, that's not possible in Python without violating the best practices for using modules. That's why I'd like to invite you again to elaborate on that. Please see http://www.chiark.greenend.org.uk/~sgtatham/bugs.html for inspiration.

Doing that is important because what you're asking us is to break existing public interface of the module for each and every MicroPython user. That's possible, but there should be very good, elaborate reasons for doing so.

peterhinch, 2017-03-18:
Clue:
```python
import nptime
```
or
```python
from nptime import time, settime
```
The ``from module import *`` syntax should normally be avoided because of its potential for this type of namespace conflict.

pfalcon, 2017-05-06:
So, hopefully the cause of the initial issue is reduced to not very correct import usage.

As for the subject of the ticket, while naming may be indeed not ideal, it's in production for a long enough time. So, any change proposal should include consideration how to really improve consistency of the overall module interface, not just suggest random change here and there which will only frustrate people. So, let me close this for now, any such proposals are welcome to be submitted separately (but likely will need a forum discussion too).

closed 2017-05-06 by pfalcon