micropython/micropython #2907: py/objboundmeth.c: bound method requires unary op __hash__

ghseb, 2017-02-24:
Prerequisite:
```
class C:
    def foo(self):
        pass
c = C()
```
C Python:
```
hash(c.foo)
32493
```

Micropython:
```
hash(c.foo)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported type for __hash__: 'bound_method'
```

I would propose to add **mp_generic_unary_op** as **unary_op**.

This would allow storage of boundmethods in sets, etc.

Sebastian

dpgeorge, 2017-02-25:
> I would propose to add mp_generic_unary_op as unary_op.

Yes this would be ok, it costs nothing.

> This would allow storage of boundmethods in sets, etc.

That's not really a sensible thing to do because `c.foo is c.foo` is a falsehood.  So:
```
>>> print({c.foo, c.foo})
{<bound method C.foo of <__main__.C object at 0x7fa9e5653b00>>, <bound method C.foo of <__main__.C object at 0x7fa9e5653ac8>>}
```

peterhinch, 2017-02-25:
Perhaps I'm missing something here but it works in CPython:
```python
class C():
     def foo(self):
         print('foo')
```
``` 
>>> c = C()
>>> s = {c.foo, c.foo}
>>> len(s)
1
>>> s
{<bound method C.foo of <__main__.C object at 0x7fa521b86438>>}
>>> 
```

dhylands, 2017-02-25:
And in CPython:
```
>>> c.foo == c.foo
True
>>> c.foo is c.foo
False
```

dpgeorge, 2017-02-28:
It seems with my example above (`{c.foo, c.foo}`) I used a different instance `c` for the 2 elemets, like: `{C().foo, C().foo}`.

So the only way to make hashing of bound methods the same as CPython is to implement a hash function that takes into account the instance and the method, something like `instance_ptr ^ method_ptr`.

Also, this discussion shows that uPy gives the wrong result for equality of bound methods: they should be equal if the instance and method are equal.

closed 2017-08-28 by ghseb