micropython/micropython #2908: unix port: Get x86_64 binary under 400000 bytes.

pfalcon, 2017-02-24:
Time to do some clean up and optimizations.

pfalcon, 2017-02-24:
0982884655a2b8619f8334b5fdb9f2e1f59670b4

ARF1, 2017-02-24:
To avoid "MicroPython is big bloat" impression, might it not make sense to implement a [custom build service like they have for nodemcu](https://nodemcu-build.com/)?

It seems to me that this would allow more freedom to add nice-to-have features while still potentially providing very compact builds.

dpgeorge, 2017-02-24:
It seems that 400000 bytes is a rather arbitrary metric to go by.  With ports like microbit and 512k-esp8266 modules there are hard size constraints, and the features that are enabled are because of the type of hardware and intended users.  Eg microbit has a speech synthesiser and a small non-fatfs filesystem, but not complex numbers; and esp8266 has socket and ussl modules because its main hardware feature is wifi.

But the unix port has no real size constraint and the features it wants are not well defined (eg does it need btree?).

Anyway, the simplest thing to do to reduce unix binary size in disable assertions.  That gets it down to 378375 bytes for me.

peterhinch, 2017-02-25:
One use for the Unix port is as a fast means of testing MicroPython code before transferring to the target platform. For example you might want to test with large datasets which would be impractical on the target. On this basis I'd say include as much as possible; definitely complex numbers and btree as features typical of types of code you might want to test.

pfalcon, 2017-02-25:
> But the unix port has no real size constraint

It has, as explained in https://github.com/micropython/micropython/wiki/ContributorGuidelines

> and the features it wants are not well defined 

Also pretty well defined - we want it to provide most of the features uPy provides (while not growing further ;-) ).

> (eg does it need btree?).

Yep, as was discussed. We want to provide solution for full-stack development, which requires DB support, and we want that to be fully portable from the MCUs up, hence btree module.

> It seems that 400000 bytes is a rather arbitrary metric to go by.

It should be pretty obvious - it's now ~408K, and as a balancing measure against patches which propose to bloat it further, here's alternative proposal.

pfalcon, 2017-02-25:
> disable assertions. That gets it down to 378375 bytes for me.

Well, maybe we should do that soon, after making another pass on asserts to make sure we converted relevant ones to exceptions.

But my original motivation is again, to ring a bell to all the new folks who exhibit following pattern: "What a *little* cute thing! Let's do something to it!" And that "something" turns out to be "grow it bigger, bigger, BIGGER!" Sorry guys, don't steal the fun from the future generations - they want to discover this cute *little* thing too. In that regard, I think that codesize-grow check for the minimal port was a helpful thing, but it doesn't fail enough patches, so I'd be looking forward to enable it for unix port too.

dpgeorge, 2017-02-27:
> It has, as explained in https://github.com/micropython/micropython/wiki/ContributorGuidelines

Yes, but it's difficult to measure code size when dynamically linked libraries are in play.  Eg how big is CPython3 really?  Why don't we disable axtls and use a dynamic library instead?

> Also pretty well defined - we want it to provide most of the features uPy provides ... and we want that to be fully portable from the MCUs up, hence btree module.

How about enabling framebuf module then?  Otherwise one has to run the coverage build to test scripts that use framebuf.

pfalcon, 2017-02-27:
> Why don't we disable axtls and use a dynamic library instead?

So, we're looking in the opposite direction - not rely on any dynamic library and have a fully static build. libffi is the last dependency which is scheduled to go: https://github.com/micropython/micropython/issues/2191 . The reason for that is well known: there's no guarantee that there's a needed dynamic library on the target system, and no guarantee that it can be built easily, or built at all. (Try your router, try dozen of other embedded Linux systems). And "MicroPython runs everywhere".

> How about enabling framebuf module then?

Well, that's not how "full stack" idea works. It should provide a solution for data storage and UI. We already make a compromise on data storage part by not providing a full SQL database, and there's one default way to make UIs - with HTML over network. Besides, does framebuf support linuxfb, OpenGL, OpenGLES, OpenVG? No? That's what I thought ;-).

> Otherwise one has to run the coverage build to test scripts that use framebuf.

Yes, we came to such solution, and it works pretty well. Can start to think about offloading the production build from assertions, etc. indeed.

dpgeorge, 2017-02-28:
> So, we're looking in the opposite direction - not rely on any dynamic library and have a fully static build.

So then people should realise this when complaining that uPy is "bloated".

> Well, that's not how "full stack" idea works. It should provide a solution for data storage and UI.

A framebuf is exactly what you need for a UI on a device that does not have a network!  So it's needed in the default build.

> Besides, does framebuf support linuxfb, OpenGL, OpenGLES, OpenVG?

Of course it does.  It also supports ASCII.  You just copy the data out to the hardware.

pfalcon, 2017-02-28:
> A framebuf is exactly what you need for a UI on a device that does not have a network!

But hardware which runs unix ports has a network  in like 95% of cases, while headless in probably 50% of cases.

> So it's needed in the default build.

Well, this ticket discusses how to decrease the size, not how to increase it ;-).

dpgeorge, 2017-07-11:
@pfalcon I'd like to close this issue as I don't see it ever being resolved to any level of satisfaction. Currently unix can be built without debugging to get it (a lot) under 400,000.  The 64 and 32 bit version have different sizes (32 bit is 360k with debugging and 340k without debugging) and so does the nanboxing version.  In the future adding support for things like VFS posix and mbedtls (yes we need it, or hope axtls adds features very soon) are going to change the size such that 400,000 doesn't have a real meaning.

pfalcon, 2017-07-11:
Why this ticket is worse than 300 other open? It waits its turn, as many other. There're action items here: disable assertions, establish Travis code size check. 400000 is just a figure to keep the title short, the real meaning of this ticket is "keep uPy from growing, find ways to shrink it instead". That's not really attainable, and it'll keep growing, but tickets like this should remind that it should be well-considered exceptions, not extensive/explosive growth.

dpgeorge, 2017-07-11:
> Why this ticket is worse than 300 other open?

It's distracting becuse, as I said, there's no clear goal (especially if 400000 is just a figure to keep the title short!).  And 400000 is anyway already met by 32-bit port with debug enabled.

> the real meaning of this ticket is "keep uPy from growing, find ways to shrink it instead"

It shouldn't be a ticket because it can never be closed.

pfalcon, 2017-07-11:
Tickets should be a reminders of work to do, and there's work to do. The title clearly states it's for x86_64 port. Of course, if you really want to close it, feel free to it, I don't see how that helps with the issues we discuss.

stinos, 2017-07-11:
> "keep uPy from growing, find ways to shrink it instead"

Isn't that sort of covered by the guidelines, the code size check in the build process, and (less visibly, but only at first sight) because it has been mentioned often in the issue comments?

dpgeorge, 2017-07-12:
> Tickets should be a reminders of work to do, and there's work to do

I'd like individual tickets to have definite goals that can be realistically reached, rather than a vauge description like "make MicroPython better", which of course everyone is trying to do.

> the title clearly states it's for x86_64 port.

Yes, sorry, I missed that.

So, assertions for unix build were disabled in 1e6fd9f2b4072873f5d6846b19b2ef0ccc5e4e52, and 64-bit executable is now 377000.

pfalcon, 2017-07-12:
Err, but did you go thru assertions one by one to check which need to be converted e.g. to exceptions? Need to do this was the reason why this idea waited in queue.

pfalcon, 2017-07-12:
>> "keep uPy from growing, find ways to shrink it instead"

> Isn't that sort of covered by the guidelines, the code size check in the build process, and (less visibly, but only at first sight) because it has been mentioned often in the issue comments?

Yep, and a lot people still manage to miss it, so this was made to ease the reference to it. And the code size check exists only for minimal build, this ticket proposes to add it for unix port too (the obvious drawback is that 95% of PRs will fail).

dpgeorge, 2017-07-14:
> Err, but did you go thru assertions one by one to check which need to be converted e.g. to exceptions?

I checked most of them, and all I found was d0db93cf1ff8eb8b2db63da07dd14d86577eebd1.  Likely there are a few assertions remaining that should be converted.

dpgeorge, 2017-07-17:
In the end binary size seems to be meaningless on unix, since the latest gcc v7.1.1 compiles the unix port to 43300 bytes, up from 377000 (64-bit build).

pfalcon, 2017-07-17:
Of course, there's a meaning, it's just there's bloat everywhere, rampaging. And we should not succumb to it.

dpgeorge, 2017-07-18:
More info on the "latest" gcc version: it's Arch Linux's gcc-multilib 7.1.1-4, built on 1/07/2017.  It seems to use position independent code by default, for both 32-bit and 64-bit x86 archs.  Code size is increased by about 15%.  The previous Arch Linux version (7.1.1-3) doesn't have this issue.

pfalcon, 2017-10-31:
I'd like to reopen this issue, to remind everyone (including myself) that the code size growth, and without reflecting on that and controlling that (the purpose of this ticket), the growth will uncontrollable. So, some thought and action should be put into ways to compensate the growth. (E.g. I opened https://github.com/micropython/micropython/issues/3319 and dumping stuff there).

dpgeorge, 2020-06-09:
I'll close this issue again because I don't see how it's useful to keep it open.

closed 2020-06-09 by dpgeorge