micropython/micropython #2914: lcd160cr jpeg method appears not to work

peterhinch, 2017-02-26:
This arose from [this thread](http://forum.micropython.org/viewtopic.php?f=6&t=3074&p=18147#p18147)
The following code erases the screen but fails to render the jpeg
```python
from lcd160cr import LCD160CR
buf = bytearray(10000)
with open('/sd/test.jpg', 'rb') as f:
    f.readinto(buf)
lcd = LCD160CR('Y')
lcd.set_pen(0, 0)
lcd.erase()
lcd.set_pos(0, 0)
lcd.jpeg(buf)
```
The 9590 byte jpeg file may be found [here](http://hinch.me.uk/lcd/test.jpg). It's a 53x53 pixel file created from a photo using Gimp.

chuckbook, 2017-02-26:
The LCD160CR does not support EXIF decorators. An easy way to strip jpeg files is to use jpegtran. Another potential problem might be the missing length overflow check in lcd160cr.py.
An extra check in lpc160cr.py might be helpful.
```
    def jpeg_start(self, l):
        if l > 0xffff:
            raise ValueError('length must be less than 65536')
        self.oflush()
        self._fcmd2('<BBH', 0x6a, l)

```

kamikaze, 2017-02-26:
```
$ file frames/nuclear.jpg 
frames/nuclear.jpg: JPEG image data, JFIF standard 1.01, resolution (DPI), density 72x72, segment length 16, baseline, precision 8, 128x160, frames 3
$ ls -l frames/nuclear.jpg 
-rw-r--r-- 1 kamikaze kamikaze 23551 feb 25 22:16 frames/nuclear.jpg
```

I've saved my JPEG file in Gimp and unchecked EXIF option.

peterhinch, 2017-02-27:
@chuckbook Thank you, that fixed it.

closed 2017-02-27 by peterhinch

dpgeorge, 2017-02-27:
In 528aeb3bf3e42c235f5822de04b16778d284a7df a check was added that the size of the jpeg is 65535 or less.

chuckbook, 2017-02-27:
@kamikaze Your jpeg file still uses unsupported VCrCb sub sampling (4:4:0 instead of required 4:2:2). After an open & save in gimp (4:2:2 subsampling, not progressive, no thumbnails etc.) it works fine.