micropython/micropython #2916: NameError: name 'i' is not defined when using threads

crasu, 2017-02-26:
Hi,

I am playing around with the _thread module. I am having the problems that some variables are suddenly vanishing.
```python
import _thread
import utime

i=0
def run():
    global i
    while True:
        i = (i + 1) % 100
    print("terminating")

def start():
    _thread.start_new_thread(run, ())
    print("sleeping")
    utime.sleep(3)
    print("wakeup")
```
When I run this I get:
```python
>>> bug.start()
sleeping
wakeup
>>> Unhandled exception in thread started by <function run at 0x3ffe2740>
Traceback (most recent call last):
  File "bug.py", line 8, in run
NameError: name 'i' is not defined
```
Is this expected? Seems surprising to me.

I am using micropython 1.8.7 on the wipy 2.0.

dpgeorge, 2017-02-27:
> I am using micropython 1.8.7 on the wipy 2.0.

If you're using wipy 2.0 (an esp32 based board) then please report this bug to Pycom: https://github.com/pycom/pycom-micropython

crasu, 2017-02-27:
Done see https://github.com/pycom/pycom-micropython/issues/85

crasu, 2017-02-27:
./micropython 
MicroPython v1.8.7-332-g528aeb3 on 2017-02-27; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import bug
>>> bug.start()
sleeping
Unhandled exception in thread started by <function run at 0x88991280>
Traceback (most recent call last):
  File "bug.py", line 8, in run
NameError: name 'bug' is not defined

peterhinch, 2017-02-27:
Odd that in this instance the error message refers to 'bug' rather than 'i'.

Pasting your code at the REPL in Unix build v1.8.1-1410-g71ae3f3-dirty on 2017-02-21 it works fine:
```
>>> start()
sleeping
wakeup
>>> 
```

danicampora, 2017-02-27:
@dpgeorge I have another example that reproduces this. It uses several files, but they are really small. It's easily reproducible on the Unix port too, so it's not a problem of the ESP32 port.

Here's my tests:

**main.py**

```python
import utime, gc, _thread
import mytask, myfunc

i = dffTme = 0

thrdLock = [_thread.allocate_lock() for i in range (2)]

while 1:
    strtTme = utime.ticks_ms()
    print('main loop',i,dffTme,'ms')

    if not thrdLock[0].locked():                                              # check if task is running
        _thread.start_new_thread(mytask.task,(thrdLock, 0, i, 10))            # start another run
    else:
        print("Thread busy")

    myfunc.funct(i)
    if i < 998: 
        i += 1
    else: 
        i = 0

    dffTme = utime.ticks_diff(strtTme,utime.ticks_ms())
    utime.sleep_ms(741 - dffTme)

```

**myfunc.py**

```python
ba=bytearray(1000)

def funct (i):
    global ba
    for b in range(1000):
        ba[b] = i^b
```

**mytask.py**

```python
import utime

def task (thLck, thId, cnt, dly):
    with thLck[thId]:
        for i in range(cnt+1):
            print(i)
            utime.sleep_ms(dly)
```

Then, if you run **main.py** you'll get:

```
main loop 0 0 ms
0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/danicampora/Code/Pycom/Shaun/main.py", line 17, in <module>
  File "/Users/danicampora/Code/Pycom/Shaun/myfunc.py", line 6, in funct
NameError: name 'ba' is not defined
```

And if you run it enough times, you might also get an error like this:

``
main loop 0 0 ms
0
main loop 1 -1 ms
0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/danicampora/Code/Pycom/Shaun/main.py", line 17, in <module>
  File "/Users/danicampora/Code/Pycom/Shaun/myfunc.py", line 6, in funct
NameError: name 'ba' is not defined
>>> 1
Unhandled exception in thread started by <function task at 0xfd7fac0>
Traceback (most recent call last):
  File "/Users/danicampora/Code/Pycom/Shaun/mytask.py", line 7, in task
  File "/Users/danicampora/Code/Pycom/Shaun/mytask.py", line 7, in task
NameError: name 'utime' is not defined
```

peterhinch, 2017-02-27:
In my post above I spoke too soon. When I closed upython I got a segfault.

dpgeorge, 2017-02-27:
Thanks guys, I see the problem: globals and locals are considered shared across all threads instead of being per-thread.

danicampora, 2017-02-27:
@dpgeorge thanks for the quick feedback!

danicampora, 2017-02-27:
> In my post above I spoke too soon. When I closed upython I got a segfault.

This is a different issue IMO. With @crasu's example, the thread running on  the function ```run()``` executes forever and there's no way to stop it. So for instance when I try to do a soft reset with Ctlr+D I get a core dump on the ESP32.

@dpgeorge don't we need a way to kill all threads during soft reset? Perhaps we can implement something that also sends the keyboard interrupt to all threads when Ctrl+C is pressed. Views?

dpgeorge, 2017-02-27:
> When I closed upython I got a segfault.

That also happens on CPython sometimes (maybe not for this example, but there's definitely cases where you can crash CPython if you don't close all threads before exiting).

> don't we need a way to kill all threads during soft reset? Perhaps we can implement something that also sends the keyboard interrupt to all threads when Ctrl+C is pressed. Views?

stmhal now has support for multithreading (but it's still in testing, not enabled by default) and it cleans up all threads when it does a soft reset.  To do it with FreeRTOS is probably just a case of going through all active tasks and killing them.

As for ctrl-c: per CPython's behaviour it should only propagate to one of the threads.

PR #2918 (2017-03-06): py: Fix access of locals/globals within a thread.
Should fix #2916.
closed 2017-03-06 by dpgeorge