micropython/micropython #2919: str.format appears to check for ...IS_TYPE not ...IS_SUBCLASS when handling dict

cefn, 2017-02-27:
I created a 'Resolver', as a subclass of dict, which intervenes in ```__getitem__``` and does str.format expansion, recursively, against its own keys. 

For example, the attached zip contains python code which implements Resolver and configures the following str.format expansion...

```
r = Resolver()
r["biter"]="man"
r["bitee"]="dog"
r["headline"]="{biter} bites {bitee}"
r["article"]="{headline}\nOn Tuesday a {biter} bit a {bitee} right on the"
print(r["article"])
```

I can run this routine in python3 like this...

```
$ python3 format_dict_type_checking.py 
man bites dog
On Tuesday a man bit a dog right on the
```

...but if it runs in micropython 1.8.7 I get the following...

```
$ micropython format_dict_type_checking.py 
Traceback (most recent call last):
  File "format_dict_type_checking.py", line 28, in <module>
  File "format_dict_type_checking.py", line 14, in __getitem__
TypeError: argument should be a 'dict' not a 'Resolver'
```

Is it possible that the type checking within str.format (possibly in https://github.com/micropython/micropython/blob/v1.8.7/py/objstr.c though I couldn't pin down where) could be loosened to permit any subclasses of dict to satisfy str.format or is this a fundamental technical and optimisation limitation that I'm running into?

[format_dict_type_checking.py.zip](https://github.com/micropython/micropython/files/804329/format_dict_type_checking.py.zip)

pfalcon, 2017-02-27:
> str.format appears to check for ...IS_TYPE not ...IS_SUBCLASS when handling dict

And not only str.format, but many other functions in uPy.

> or is this a fundamental technical and optimisation limitation that I'm running into?

Yes. In MicroPython, subclassing builtin types is not recommended. Basic support is there, but if you go beyond the most basic things, you'll hit a gazillion cases like above (which can't be all fixed except by compromising MicroPython's basic principles), so won't want to subclass yourself.

cefn, 2017-02-27:
Thanks @pfalcon. Understood and I'll look for another way.

closed 2017-02-27 by cefn