micropython/micropython #2920: RFC: filesystem layout, VFS and user mounting

dpgeorge, 2017-02-28:
As brought up in #2850, there has been a change to esp8266's filesystem.  Previous to the change the root dir was the one-and-only filesystem, being a FatFS filesystem backed by the flash.  After the change, the root dir is a virtual filesystem upon which you can mount an abritary number of filesystems, of whatever type you like (included one implemented purely in Python).

This change makes esp8266 the same as pyboard/stmhal: the flash is mounted at /flash.  And you can mount an SD card at /sd.

If you unmount everything (eg os.umount('/flash')) you still have the root dir, and can then mount filesystems there again (eg os.mount(flashbdev, '/flash2')).

dpgeorge, 2017-02-28:
From comment https://github.com/micropython/micropython/issues/2850#issuecomment-278992804

> Or maybe it's easier after all to just follow simple and well-known world model where you have a real filesystem on / as the most basic thing?

The root dir at / is a virtual filesystem, and is independent of any particular type (eg FatFS).  So ports that don't want to use FatFs can still have a filesystem and can still mount things there.  Making the root backed by a particular filesystem (like FatFs) complicates things because then you have to check for mount points within that FS.  Also, if you unmount the root, what's left?  How/where do you remount anything?

> Also: was it worth the trouble to implement a real VFS system, like some big OS, and then fall short at oversimplistic mount table implementation?

It was implemented in the simplest way to allow mounting of multiple filesystems.

> What's the concern anyway?

Adding unnecessary complexity to the mounting code.  And dealing with a non-mounted root filesystem.

pfalcon, 2017-02-28:
Good timing, I keep forgetting to post yet another argument for doing it right - for unix port, we could move POSIX filesystem support into VFS. Of course, then we'd need it to mount on /.

pfalcon, 2017-02-28:
Thanks for posting this! Let's see:

> The root dir at / is a virtual filesystem, and is independent of any particular type (eg FatFS).

Ok, great, I don't call to change that, only call to support mounting another filesystem over such "virtual root".

> Making the root backed by a particular filesystem (like FatFs) complicates things because then you have to check for mount points within that FS.

Well, mount points  remain outside any particular FS. Unless you mean possibility of a conflict, like some FS mounted on root, has a dir "/foo", then user asks to mount another FS on "/foo". Well, in 1st approximation, we don't even need to check for that, and let users do what they want. Adding a check later shouldn't be a big problem though.

>  Also, if you unmount the root, what's left? How/where do you remount anything?

pfalcon, 2017-02-28:
>> What's the concern anyway?

> Adding unnecessary complexity to the mounting code. And dealing with a non-mounted root filesystem.

So the compromise change I call for (vs complete POSIX mounting support) is that a VFS can be mounted on a root / , in addition to on an one-level directory below it, like /foo. I appreciate that this already adds more complexity to mounting code, but a code is supposed to do useful things, and if it's too simple to surprise humans with not following well-known patterns, that's not good.

The needed changes should be pretty bearable: /, if mounted, should be at the well-known location in the mounts list. Apparently, the should be end of list - if nothing matched before it, root is fallback. And ilistdir() function, if called on "/", should chain iteration of mount points to delegated call to ilistdir() of the real file system. Those are all the "algorithm-level" changes I can envision, though probably more checks here and there may need to be added.

dpgeorge, 2017-03-01:
> I keep forgetting to post yet another argument for doing it right

IMO I already did it the "right" way based on uPy principles: it's the simplest implementation that works and gives users the features they need (mounting multiple devices, without restriction on the filesystem type).  Also, pyboard has had the directory structure of (/flash, /sd) for many years and people seem just fine with it.

> So the compromise change I call for (vs complete POSIX mounting support) is that a VFS can be mounted on a root / , in addition to on an one-level directory below it, like /foo. I appreciate that this already adds more complexity to mounting code

Yes it does.  And it's already complex/big enough that I don't really want to increase it anymore.

> a code is supposed to do useful things, and if it's too simple to surprise humans with not following well-known patterns, that's not good.

As I say above it already does everything you need, and no more.  And what's surprising about it how it is now?  That you can't mkdir('/lib')?  But you can't do that on unix either without being root.  You don't usually play with the root of the filesystem like that.

> And ilistdir() function, if called on "/", should chain iteration of mount points to delegated call to ilistdir() of the real file system.

I think that's going to require a bit of work and extra code.

dpgeorge, 2017-03-01:
Apart from implementation issues, there's the question of why?  I propose to have the standard directory structure as stmhal and cc3200 have it: /flash for the internal flash, /sd for a mounted sd card, etc.

pfalcon, 2017-03-01:
> Also, pyboard has had the directory structure of (/flash, /sd) for many years and people seem just fine with it.

Yes, people are fine with it - on pyboard.

> Apart from implementation issues, there's the question of why? I propose to have the standard directory structure as stmhal and cc3200 have it: /flash for the internal flash, /sd for a mounted sd card, etc.

Yes, that "standard directory structure" doesn't look too bad for cute embedded hacks like stmhal and cc3200. But MicroPython is much bigger than that, and as soon as we consider a bit more involved hardware, that scheme becomes a weird idiosyncrasy given people unexpected surprises. For example, consider a hardware which normally runs Linux and on which instead MicroPython started - but there will be no end of questions why it's done that way that people can't have /lib, or /usr, or /etc, or /home, unless they mount a separate partition on that. There's no good answer for why it was done like that, and something "ugh, we should for the simplest possible solution, so we didn't think about" doesn't cut (and something like "we shoot for the simplest possible solution, so yeah, get a separate partition ready for each of them" is even worse).

And how all this maps to unix port? I already gave a hint about that above. And I called many times to think how some functionality applies to the unix port, and if this thinking wasn't done, or there's no good answer, then it means that design wasn't done right. This applies to this, this applies to framebuffer module, this applies to I2C/SPI API, etc. So, for unix, we could have native POSIX VFS mounted on /, and then still allow user to mount other filesystem types on uPy side, instead of having a hack of moduos_vfs.

pfalcon, 2017-03-01:
> Yes it does. And it's already complex/big enough that I don't really want to increase it anymore.

Sorry, all these months I wonder, how you would implement the mounting table: would you use mp_map_t but iterate it to find the longest match, or will sort mount entries - of course assuming you would go for arbitrary mounting support. And you say that having a flat list, instead of a tree, is complex. Well, it's not complex. Or maybe it's "complex", but not really complex as other things. And it's definitely too simplistic for many (most) real-world cases.

> I think that's going to require a bit of work and extra code.

I agree with assessment that it will be "a bit". Shouldn't be too much. If you don't feel like doing it, I can volunteer for that, but I'm busy these weeks, and if we agree that it needs fix at all, then https://github.com/micropython/micropython/issues/2850 is a noticeable regression, so fix to it blocks the next release.

And well, I genuinely feel your point and agree that the current implementation is the simplest possible, and that's cool. But it's the last mile problem - by not going that last mile, we're cutting uPy's VFS expressiveness in half. My reaction in #2850 is a genuine reaction - I couldn't even think that mounts have changed behind my feet, I thought that paths were incorrectly updated. More people would have such reaction if everything goes by the plan and uPy gets used in more places.

deshipu, 2017-03-23:
If I remember correctly, the original UNIX systems also didn't allow files in the root -- a vestige of that is the separation of `/bin` and `/usr/bin` -- they were separate partitions, and once they ran out of room in `/bin` they created a second one in `/usr`, which originally was supposed to hold user home directories.

pfalcon, 2017-05-06:
@dpgeorge : Should we keep this open for discussion on future points in elaborating VFS API? If so, let's not forget about it and not spread the discussion around ;-).

dpgeorge, 2017-05-06:
> Should we keep this open for discussion on future points in elaborating VFS API?

I'd rather close it and discuss individual points in individual tickets, otherwise we end up with 100's of comments in an issue and it becomes impossible to 1) follow a single train of discussion; 2) close the issue because there's no well defined way to finish the discussion.

pfalcon, 2017-05-06:
Thanks, closing.

closed 2017-05-06 by pfalcon