micropython/micropython #2924: tests: Switch to use "raise SystemExit" for exit

pfalcon, 2017-03-02:
A fresh thought: we currently use:

~~~
import sys
sys.exit()
~~~

in tests to stop its execution. Somewhere deep, it always bothered me that Python doesn't have a "builtin" way to terminate a program. Well, it has - it's "raise SystemExit". That's a bit of paradigm shift, because my C-think makes me think "well, sys.exit() actually *could* be implemented as _exit()", but in Python, sys.exit() is definitely nothing but a facade to "raise SystemExit".

Converting tests to such will make them independent of "sys" existence (but of course will make them dependent on SystemExit existence).

dpgeorge, 2017-03-02:
Sounds good to me.

> Converting tests to such will make them independent of "sys" existence (but of course will make them dependent on SystemExit existence).

Actually this helps with portability of scripts: currently you can disable the sys module and so certain ports may not have it.  But you can't disable SystemExit exception, it's included in all builds.

And how about going further and raising a certain magic number `raise SystemExit(-123)` instead of printing SKIP and raising 0?

dpgeorge, 2017-06-07:
I think it's a good idea to use SystemExit, and to gradually change the tests.

pfalcon, 2017-06-07:
Sure. The culprit in qemu-arm which uses adhoc method for running tests, which will need to be investigated and patched, so far nobody has got to that.

dpgeorge, 2017-06-07:
> The culprit in qemu-arm which uses adhoc method for running tests

I don't follow... `raise SystemExit` and `sys.exit()` are equivalent, if the latter works then the former should.  And indeed it does on qemu-arm running a test that should be skipped (basics/parse.py).

dpgeorge, 2017-06-08:
Ok, tests/floats/* were converted to use "raise SystemExit" in fde54350a8d740308d12c2ea4c65fa5bea4f186a.  It seems to work fine.

pfalcon, 2017-06-11:
Now fully converted.

closed 2017-06-11 by pfalcon