micropython/micropython #2926: Interface for tokenize module

ARF1, 2017-03-02:
Since micropython presumably already has a built-in lexer, how difficult / size-costly would it be to expose this in a way suitable to allow the implementation of the [tokenize](https://docs.python.org/3/library/tokenize.html) module from the python standard library for micropython?

The reason for my request is as follows:
I am still somewhat disappointed that I cannot use `help(my_func)` to be reminded of the correct way to use a function. And while clearly the CPython approach to the function help functionality is unsuitable for micropython, the tokenize module would allow a pure-python implementation that mimics this behaviour by parsing the plain-text code.

dpgeorge, 2017-03-08:
Note that the tokenize module is implemented in pure Python, so you could just port it to uPy.

Otherwise, if you really want to expose the lexer, just wrap mp_lexer_to_next and return a tuple of (lex->tok_line, lex->tok_column, lex->tok_kind, lex->vstr).

pfalcon, 2017-05-01:
Here's an idea (for contributors): Implement such a module, but enable it only in mpy-cross, and then gradually move it from fixed-function approach implemented in C, to a Python driver which drives parsing/compilation, then optimization, then saving. That for example would potentially allow to address cases like https://github.com/micropython/micropython/issues/3054, and much more.

pfalcon, 2018-12-26:
> Implement such a module, but enable it only in mpy-cross

I now started to look into such stuff myself, in the context of https://github.com/micropython/micropython/pull/4309. tokenize is too boring though, I started straight with ast. But when I coded a module skeleton, it occurred to me that MicroPython doesn't really have AST! It has only parse tree, and that's much more verbose and ugly to work with. So, I put it aside, and it only emphasizes what was said before:

> to a Python driver which drives parsing/compilation, then optimization, then saving

Except it shouldn't be "Python driver", it should be written completely in Python at all.

jonnor, 2024-07-19:
Is this still relevant? It does not seem to block anyone from using MicroPython, and it sounds like it is possible to write the functionality in plain Python - ie nothing needed in MicroPython itself. Maybe it can be closed?

closed 2024-07-20 by dpgeorge