micropython/micropython #2928: Messed up traceback for caught exceptions

ARF1, 2017-03-02:
For exceptions that were caught, the frame ordering in a traceback appears to be erroneous:

```
MicroPython v1.8.7 on 2017-03-02; win32 version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== def some_fn():
===     foo()
===
=== def foo():
===     bar()
===
=== def bar():
===     raise RuntimeError('something')
===
=== def some_other_fn():
===     pass
===
=== try:
===     some_fn()
=== except Exception as e:
===     # do something
===     raise
=== finally:
===     some_other_fn()
===
===
Traceback (most recent call last):
  File "<stdin>", line 19, in <module>
  File "<stdin>", line 17, in <module>
  File "<stdin>", line 17, in <module>
  File "<stdin>", line 14, in <module>
  File "<stdin>", line 2, in some_fn
  File "<stdin>", line 5, in foo
  File "<stdin>", line 8, in bar
RuntimeError: something
>>>
```

There appear various things weird to me:

1. The order of the entries for lines 14-19 is reversed with respect to the order of execution.
2. The entry for line 17 is duplicated even though it is executed only once.

I believe the correct traceback should be:

```
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
  File "<stdin>", line 2, in some_fn
  File "<stdin>", line 5, in foo
  File "<stdin>", line 8, in bar
  File "<stdin>", line 17, in <module>
  File "<stdin>", line 19, in <module>
RuntimeError: something
```

ARF1, 2017-03-02:
Just for reference, on CPython the traceback is as follows:

```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "<stdin>", line 2, in some_fn
  File "<stdin>", line 2, in foo
  File "<stdin>", line 2, in bar
RuntimeError: something
```

dpgeorge, 2017-03-03:
The reason for the extra lines (at least line 19) is that a traceback is being added when the exception is re-raised after executing the finally clause.

dpgeorge, 2019-08-21:
See #5032 for a fix.

dpgeorge, 2019-08-28:
Fixed by 16f6169c88a5fd93b151149e561d177557fcc760 and 08c1fe556915c912598d5a5d5db0f5942f3a333d

closed 2019-08-28 by dpgeorge