micropython/micropython #2930: esp8266: socket.connect in non-blocking mode returns ETIMEDOUT, not EINPROGRESS

dpgeorge, 2017-03-03:
The following code behaves differently on CPy/unix uPy vs esp8266:
```python
import socket
s = socket.socket()
s.setblocking(False)
s.connect(socket.getaddrinfo('micropython.org', 80)[0][-1])
```
esp8266 raises ETIMEDOUT, whereas CPy and unix uPy raise EINPROGRESS.  uasyncio depends on the error being EINPROGRESS.

pfalcon, 2017-06-03:
Fixed in a0dbbbebb8c0286e00ae06751b0173cbca4ec801. example_http_client.py from uasyncio tested to work as expected.

Not closing to serve as a reminder for need for automated tests...

pfalcon, 2017-06-04:
More cleanups for error codes in modlwip: 5da8de2b66d3f43107e1e745afa9bb6a4bf601eb

pfalcon, 2017-06-04:
From https://github.com/micropython/micropython/issues/2947:

> Note that if the socket is in a blocking mode with a finite timeout (eg s.settimeout(1)) then ETIMEDOUT is the correct error code to raise (matches CPy's socket.timeout exception for this case).

It's unclear how a timeout case would apply to connect(). The above patch just switches error code to EINPROGRESS (I did a quick look at CPython code, but didn't see much, and decided not to bother further, because there're really many different socket timeouts, but we can set only one, so there will be more issues, which can be looked at when reported).

dpgeorge, 2017-06-21:
A test for this was added in 458cbacb8f62ad0f33024360af61fb797530c65b

closed 2017-06-21 by dpgeorge