micropython/micropython #2931: esp8266: uselect.poll() should return ready-for-reading when peer closes connection

dpgeorge, 2017-03-03:
The following code runs fine on unix uPy (and CPy with appropriate changes to module names) but not on esp8266:
```python
import socket, uselect
s = socket.socket()
s.connect(socket.getaddrinfo('google.com', 80)[0][-1])
s.send(b'GET /\r\n\r\n')
s = s.makefile('rb')
poll = uselect.poll()
poll.register(s, uselect.POLLIN)
while 1:
    poll.poll() # wait for something to read
    l = s.readline()
    if len(l) == 0:
        break
    print(l)
poll.poll() # peer has closed the connection and this should return immediately
s.close()
```
On esp8266 the loop never finishes.  This is because the lwip code considers "peer closed the connection" as a HUP poll event, not a RD event.

pfalcon, 2017-04-29:
Well, so this appears to be another POSIX drama: http://www.greenend.org.uk/rjk/tech/poll.html . The most creative outcome from the reading it I can come up with is: POLLHUP should be set if the underlying network connection went down. This what directly corresponds to "modem disconnect", etc. It also makes good sense to report it, because otherwise, a connection may go down, peer may notice a timeout, send us bye, which we never receive, at which point connection goes up, and we never receive anything on that socket forever (won't ever return from poll.poll(-1) call). So, it makes sense to let clients no that something may have gone wrong underlyingly, and then the only way to do that is POLLHUP (such condition is not error per se - classical TCP may work across link layer disconnects). Note that POLLHUP is "unsolicited" return code, you aren't supposed to ask for it ("This flag is only valid in the revents bitmask; it is ignored in the events member."), and you may receive it at all times.

So, we won't be implementing such POLLHUP handling here, but that's enough to decide that "peer closed" event should lead to POLLIN and nothing else.

pfalcon, 2017-04-29:
Another interesting case is whether POLLOUT should be set for peer closed socket? Common sense would say "no, how can closed socket be writable?". But it depends on the exact interpretation of what POLLOUT means, and it means: "Writing now will not block." And that's exactly true for closed socket - write won't block, it will error out. Not retuning POLLOUT for closed socket would mean that POLLOUT poll on a closed socket will hang forever. Let's hope that current implementation in modlwip is good enough in that regard.

pfalcon, 2017-04-29:
> Let's hope that current implementation in modlwip is good enough in that regard.

Actually no, I'm fixing that too in one pass.

dpgeorge, 2017-05-06:
I can confirm that this is now fixed on esp8266.

closed 2017-05-06 by dpgeorge