micropython/micropython #2936: Main.py should be unbreakable

Cediddi, 2017-03-04:
There's a toolkit called mpy-utils, which have a fuse adapter. It provides a virtual mounting point for esp8266 or other micropython boards. By design, when you mount, it restarts the board and breaks for any possible application loops, then commences on running os.listdir and os.stat commands itself. 

The issue is, break commands run, even on boot.py, yet we want to break main.py. Is there a way to disable keyboardinterrupt being raised? In unix, I'd register a noop function to EINTR and de-register when bootup work is done. Because micropython has no concept of unix signals, I pass this.

Second option would be this:

    flag = False
    while True:
        try:
            #do stuff
            break
        except KeyboardInterrupt:
            flag=True
    if flag:
        raise KeyboardInterrupt()

But this is an awful hack and it'd restart my bootup code again and again. 

Another option would be a "break only main" command, something like Ctrl-H to break main only. But I think this might lead to a more complicated behavior.

Final option would be waiting until boot.py finishes and then break the code, but in that case, there should be a way to get notified, a special character maybe? "\x07" a bell to notify? This behavior shouldn't break most serial interface applications.

I believe this will help developers to build a better serial communication to their devices.

dhylands, 2017-03-04:
You can call [code]pyb.USB_VCP().setinterrupt(-1)[/code] to turn off Control-C handling. It will get re-enabled as soon as main.py runs.

Cediddi, 2017-03-05:
Okay, that might work for pyboard , yet most ports including esp8266, does not have pyb module. Also you'll still consume break calls qnd ignore them, so I need to send break commands until I read the >>> prompt and switch to raw mode.

dpgeorge, 2017-03-06:
The way that pyboard/stmhal works is that after you enter raw REPL mode, if you then do a soft reset, only boot.py is executed and not main.py.  This is the behaviour pyboard.py expects (and, for example, is how the test suite can be run without running main.py for each test).  The sequence is:
```
ctrl-C ctrl-C -- stop any executing script; no-op if at the (raw) REPL already
ctrl-A -- enter raw REPL (or stay in raw REPL if already in it)
ctrl-D -- do a soft reset, executing only boot.py, then drop to raw REPL
```
From then on you can upload a script and it'll be executed.  You can upload multiple scripts, or repeat the above sequence to do a full soft reset.  In no case will main.py be executed.

dpgeorge, 2017-03-06:
See #2940 for a fix.  mpy-utils also needs to be fixed to use the ctrl sequence above for initialisation.  Currently it uses C-C-D-A but it should use C-C-A-D.

Cediddi, 2017-03-07:
Thanks for the quick fix. I'm writing a PR to mpy-utils right now. :+1:

PR #2940 (2017-03-14): esp8266: Only execute main.py if in friendly REPL mode.
This should fix #2936.
closed 2017-03-14 by dpgeorge