micropython/micropython #2938: Threading tests fail randomly on Travis CI builds

blazewicz, 2017-03-05:
Noticed on Travis CI build after applying a patch which doesn't actually change anything: https://travis-ci.org/micropython/micropython/builds/207886609

Failed here:
> FAIL  thread/thread_lock3.py

pfalcon, 2017-03-05:
That's a long known issue. Which however should be reconsidered in context of https://github.com/micropython/micropython/pull/2918

pfalcon, 2017-07-08:
We didn't have non-deterministic thread failures for a while, but here's new one: https://travis-ci.org/micropython/micropython/builds/251541078 .

~~~
1 tests failed: thread_shared1
~~~

I'm not rebuilding this build so the build stayed as is.

dpgeorge, 2017-07-25:
Another one: https://travis-ci.org/micropython/micropython/builds/257159251, again it's thread_lock3.py.

dpgeorge, 2021-04-14:
CI was moved to GitHub workflows and this issue hasn't been seen in a while, so closing.

closed 2021-04-14 by dpgeorge