micropython/micropython #2941: mpy-cross link error on macOS

BrendanSimon, 2017-03-06:
Got the following linker error on macOS building `mpy-cross` with commit `[96aa3a31]`

```
$ make -C mpy-cross
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK mpy-cross
Undefined symbols for architecture x86_64:
  "nlr_push_tail", referenced from:
      _nlr_push in nlrx64.o
     (maybe you meant: _nlr_push_tail)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [mpy-cross] Error 1
```

dpgeorge, 2017-03-07:
Thanks for the report.  @BrendanSimon can you please check if #2943 fixes the problem?

BrendanSimon, 2017-03-07:
Seems to have fixed the cross-compiler link issue, but building for the STM32F4Discovery board fails.

```
CC ../py/../lib/utils/printf.c
MPY modules/lcd160cr.py
make: *** [build-STM32F4DISC/frozen_mpy/lcd160cr.mpy] Segmentation fault: 11
```

mcauser, 2017-03-07:
Also experiencing this on OSX 10.11.6 El Capitan.
Confirmed fixed in: https://github.com/micropython/micropython/pull/2943

dpgeorge, 2017-03-08:
@BrendanSimon @mcauser can one (or both, for better stats) of you please post somewhere the disassembly of the mpy-cross/build/py/nlrx64.o file, using `objdump -xd <file>`?  I assume you both use 64-bit Mac.

mcauser, 2017-03-08:
@dpgeorge I don't have objdump, but I do have gobjdump.

Using head from master branch:
https://gist.github.com/mcauser/10e5a47f516deda75c6776990fe58c24

dpgeorge, 2017-03-08:
@mcauser thanks!  I just changed #2943 to make further fixes for Mac.  Can you please try it again?  (I push forced so you'll need to remove that branch and repull it.)

pfalcon, 2017-03-08:
>  (I push forced so you'll need to remove that branch and repull it.)

... just to use "git pull --rebase", which should be the default command to pull anyway.

mcauser, 2017-03-08:
https://gist.github.com/mcauser/8312a86491bddcaa262b978d632a1ecd

`./mpy-cross -v ../drivers/display/lcd160cr.py` no longer throws seg fault 11

I'll try building my stmhal board.
`make BOARD=VCC_GND_F407`
Success!

dpgeorge, 2017-03-08:
@BrendanSimon can you please also check the latest code in #2943?

BrendanSimon, 2017-03-08:
Yep.  The STM32F4DISCO all builds ok for me too (using arm-none-eabi-gcc v5.4.1).

dpgeorge, 2017-03-08:
The fix has been merged to master.  Thanks guys for testing.

PR #2943 (2017-03-08): py/nlrx64: Fix for Mac OS.
Should fix #2941.
closed 2017-03-08 by dpgeorge