micropython/micropython #2948: esp8266: socket blocking settings are inherited by accept() when they probably shouldn't be

dpgeorge, 2017-03-10:
Test code:
```python
import time, socket
s = socket.socket()
s.bind(socket.getaddrinfo("0.0.0.0", 8000)[0][-1])
s.setblocking(False)
s.listen(1)
while True:
    try:
        s2, _ = s.accept()
        break
    except OSError as er: 
        print(repr(er))
    time.sleep(1)
print('connection')
s2.recv(10)
```
On CPy and uPy unix, running the above code and connecting to the socket will block indefinitely at the last line, because the accepted socket `s2` defaults to blocking mode.

On esp8266 the accepted socket `s2` inherits the non-blocking behaviour from `s`.

pfalcon, 2017-06-04:
From "man accept"

       On Linux, the new socket returned by accept()  does  not  inherit  file
       status  flags such as O_NONBLOCK and O_ASYNC from the listening socket.
       This behavior differs from the canonical  BSD  sockets  implementation.
       Portable  programs  should not rely on inheritance or noninheritance of
       file status flags and always explicitly set all required flags  on  the
       socket returned from accept().

Also, see CPython bikeshed: https://bugs.python.org/issue7995 and what it led to: https://docs.python.org/dev/library/socket.html#timeouts-and-the-accept-method .

pfalcon, 2017-06-04:
So, we can leave modlwip behavior as is, knowing that MacOS/BSD/Windows will show the same behavior on Unix port. Or we can make modlwip have the same behavior as Linux, while still knowing that MacOS/BSD/Windows will show the different behavior on Unix port.

dpgeorge, 2017-06-06:
If CPython on Windows/Mac has different behaviour in this regard to running under Linux then it doesn't really matter which approach is taken.  When I wrote this issue I thought that all CPython implementations had the same non-inheriting behaviour, but added the word "probably" to the issue title because I didn't know what the "true" specification was.

Might be nice to follow traditional BSD behaviour and inherit flags.  But the main thing would be to make sure that all Python code expliticly sets blocking or non-blocking behaviour on accepted sockets.