micropython/micropython #2949: Device Crashes Catastrophically when accessing SD in certain situations

ryannathans, 2017-03-10:
Steps to reproduce

1. Make sure device has SD Card inserted.
2. Make sure device has SKIPSD on internal flash
3. Run the following in main.py on internal flash. This puts my device in a boot loop.

It seems os.listdir('/') in another module crashes the device after throwing an exception with a file on the SD card.

Some devices fail the listdir in test with memory allocation fail 7958574 bytes and some devices get stuck in a boot loop, immediately crashing and restarting without displaying an error. 

Using STM32F439 on these devices 

```
import os
import pyb

print(os.listdir('/'))

with open('/flash/test.py', 'wb') as file:
                file.write(b"import os\nos.listdir('/')\n")

if 'sd' not in os.listdir('/'):
                os.mount(pyb.SD, '/sd')
                print("os.mount() successful")

print(os.listdir('/'))

with open('/sd/test.txt', 'wb') as file:
                for i in range(10000):
                                file.write(b"This is twenty bytes.")

file = open('/sd/test.txt', 'rb')
try:
                a = file.read()
except MemoryError:
                print('Exception thrown as expected')

print(os.listdir('/'))
import test
```

dpgeorge, 2017-03-10:
Thanks for the report.  It's an issue with the GC not tracing a root pointer.  Should be fixed by 8236d18338f6d8db25dcc5e81b176b006d56f39d.  Let me know if it works for you.

dpgeorge, 2017-03-14:
I assume this was fixed.

closed 2017-03-14 by dpgeorge

ryannathans, 2017-03-14:
Haven't been able to test, just had surgery. Will reopen if there's still a problem