micropython/micropython #2952: Build error: "cp: cannot stat ‘config/upyconfig’: No such file or directory" trying to build ESP8266

jcea, 2017-03-14:
1. Download file "v1.8.7.tar.gz" from this github webpage.

2. Unpack with "tar xvfz v1.8.7.tar.gz".

3. "cd micropython-1.8.7/".

4. "make -C mpy-cross". Everything fine so far.

5. "cd esp8266".

6. "make axtls". I immediately get this fatal error:

```
$ make axtls
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
cd ../lib/axtls; cp config/upyconfig config/.config
cp: cannot stat ‘config/upyconfig’: No such file or directory
Makefile:239: recipe for target 'build/libaxtls.a' failed
make: *** [build/libaxtls.a] Error 1
```

A "find 'upyconfig'" in the source tree shows nothing. Googling is not useful either.

I was a kickstarter sponsor, help me :-)

jcea, 2017-03-14:
Verbose build shows nothing else:

```
$ make V=1 axtls
cd ../lib/axtls; cp config/upyconfig config/.config
cp: cannot stat ‘config/upyconfig’: No such file or directory
Makefile:239: recipe for target 'build/libaxtls.a' failed
make: *** [build/libaxtls.a] Error 1
```

jcea, 2017-03-14:
esp8266/README.md documents this command athe beggining of the compilation process: ```git submodule update --init```. But this is not valid for the tar.gz archive:

```
$ git submodule update --init
fatal: Not a git repository (or any parent up to mount point *****)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```

dpgeorge, 2017-03-14:
If you've got git installed then the easiest thing to do is just git-clone this repository:
```
$ git clone http://github.com/micropython/micropython.git
$ cd micropython
$ git checkout v1.8.7
$ git submodule update --init
```
Then continue from there.

jcea, 2017-03-14:
Yes, I have compiled the code cloning via GIT, but I guess that the "tar.gz" releases should work...

Great work, BTW. I am looking forward inlining ESP8266 ASM  instructions in the middle of my python code :-p

Thanks for your effort, @dpgeorge

pfalcon, 2017-03-29:
@jcea: It definitely would be nice that tarball releases. But it so happens that github creates them, and they don't work. That may be related, or not. But looks like maintainers don't have enough time to look into github issues, as we have a lot to do directly on MicroPython side. Interested parties are welcome to investigate what's wrong and either tell us how to fix it, or tell so github. In the meantime, there's an obvious workaround - to use git. So, to set expectations right, let me assign this a low-priority label. Thanks.

pfalcon, 2017-05-06:
We probably won't be able to fix this on our side, as the problem is indeed in Github's not including submodules in the release tarballs they produce. Affected parties are welcome to submit the issue report to Github and follow up on it.

Otherwise, the solution is to use git directly (highly recommended).

wgreybe, 2017-07-01:
Pfalcon, please can you elaborate with regards to what you mean: "Otherwise, the solution is to use git directly (highly recommended)."
I am getting this error as well and would like to know whether there is a work around?
Thanking you in advance.

dpgeorge, 2017-08-23:
Releases now (starting from v1.9.2) include custom-made tgz/zip files with submodules included; see https://github.com/micropython/micropython/releases

closed 2017-08-23 by dpgeorge