micropython/micropython #2957: Confusing error message for implicit bytes to str conversion

stinos, 2017-03-14:
For example:
```
>>>b'abc'.replace('a', '')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'str' object to str implicitly
```
I'm fairly sure this used to say ```can't convert 'bytes' object to str implicitly```?

dpgeorge, 2017-03-14:
Yes, the way it's handled is that there's a generic message to say it can't convert X to a str.  In this case it's trying to convert 'a' to bytes implicitly, which can't be done, and uses the generic str error mesage for that.

stinos, 2017-03-14:
Ok, just wanted to doublecheck the 'to str' part is by design even though it's converting to bytes

pfalcon, 2017-03-14:
Well, let's try to look into doing something about this before closing.

dpgeorge, 2017-03-20:
Fixed.

closed 2017-03-20 by dpgeorge