micropython/micropython #2959: compile in cygwin error

fkeujjpdc, 2017-03-14:
16062501@rd-15070201 /usr/micropython-master/micropython-master/minimal
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosit
y.
GEN build/genhdr/qstr.i.last
/usr/bin/sh: -c:行5: 语法错误: 未预期的文件结尾（Syntax error: unexpected end of file）
make: *** [build/genhdr/qstr.i.last] 错误 1（Error 1）

dpgeorge, 2017-03-14:
Please provide extra information about your issue, what OS you use, what locale, and also the output from "make V=1".

fkeujjpdc, 2017-03-15:
`$ make V=1
python ../py/makeversionhdr.py build/genhdr/mpversion.h
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
if [ "../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/                              mpprint.c ../py/unicode.c ../py/mpz.c ../py/reader.c ../py/lexer.c ../py/parse.c                               ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmbase.c                               ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/as                              marm.c ../py/asmxtensa.c ../py/emitinlinextensa.c ../py/formatfloat.c ../py/pars                              enumbase.c ../py/parsenum.c ../py/emitglue.c ../py/persistentcode.c ../py/runtim                              e.c ../py/runtime_utils.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c                               ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ..                              /py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objc                              omplex.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.                              c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ..                              /py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/ob                              jmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty                              .c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ..                              /py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrun                              icode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../p                              y/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport                              .c ../py/builtinevex.c ../py/builtinhelp.c ../py/modarray.c ../py/modbuiltins.c                               ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmat                              h.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ..                              /py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint                              .c ../py/frozenmod.c ../py/../extmod/moductypes.c ../py/../extmod/modujson.c ../                              py/../extmod/modure.c ../py/../extmod/moduzlib.c ../py/../extmod/moduheapq.c ../                              py/../extmod/modutimeq.c ../py/../extmod/moduhashlib.c ../py/../extmod/modubinas                              cii.c ../py/../extmod/virtpin.c ../py/../extmod/machine_mem.c ../py/../extmod/ma                              chine_pinbase.c ../py/../extmod/machine_signal.c ../py/../extmod/machine_pulse.c                               ../py/../extmod/machine_i2c.c ../py/../extmod/machine_spi.c ../py/../extmod/mod                              ussl_axtls.c ../py/../extmod/modussl_mbedtls.c ../py/../extmod/modurandom.c ../p                              y/../extmod/moduselect.c ../py/../extmod/modwebsocket.c ../py/../extmod/modwebre                              pl.c ../py/../extmod/modframebuf.c ../py/../extmod/vfs.c ../py/../extmod/vfs_rea                              der.c ../py/../extmod/vfs_fat.c ../py/../extmod/vfs_fat_diskio.c ../py/../extmod                              /vfs_fat_file.c ../py/../extmod/vfs_fat_misc.c ../py/../extmod/utime_mphal.c ../                              py/../extmod/uos_dupterm.c ../py/../lib/embed/abort_.c ../py/../lib/utils/printf                              .c ../py/emitnative.c" = "" ]; then \
            echo "QSTR Looks like -B used, trying to emulate"; \
            gcc -E -DNO_QSTR -DN_X64 -DN_X86 -DN_THUMB -DN_ARM -DN_XTENSA -Ibuil                              d/tmp -m32 -I../lib -I../lib/netutils -I. -I.. -I../lib/mp-readline -I../stmhal                               -Ibuild -Wall -Werror -ansi -std=gnu99  -Os -DNDEBUG ../py/mpstate.c ../py/mallo                              c.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/m                              pz.c ../py/reader.c ../py/lexer.c ../py/parse.c ../py/scope.c ../py/compile.c ..                              /py/emitcommon.c ../py/emitbc.c ../py/asmbase.c ../py/asmx64.c ../py/asmx86.c ..                              /py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/asmxtensa.c ../py/em                              itinlinextensa.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py                              /emitglue.c ../py/persistentcode.c ../py/runtime.c ../py/runtime_utils.c ../py/n                              ativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py                              /obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.                              c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdict.c ../py/ob                              jenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c                               ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlon                              g.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/ob               [... cut ...] t_misc.c ../py/../extmod/utime_mphal.c ../py/../extmod/uos_dupterm.c ../py/                              ../lib/embed/abort_.c ../py/../lib/utils/printf.c ../py/emitnative.c >build/genh                              dr/qstr.i.last; \
        fi
/usr/bin/sh: -c: line 5: syntax error: unexpected end of file
`

system:Win7 ultimate x86
gcc version: 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496] (GN
U Tools for ARM Embedded Processors)
 python: using install in windwos Python 2.7.12

dpgeorge, 2017-03-15:
It could be that the command is too long for Windows (the above one is about 8600 characters).

Try changing the file `py/mkrules.mk`, starting at line 72, to looke like the following:
```
71 
72 $(HEADER_BUILD)/qstr.i.last: $(SRC_QSTR) | $(HEADER_BUILD)/mpversion.h
73         $(ECHO) "GEN $@"
74         $(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $? >$(HEADER_BUILD)/qstr.i.last;
75 
```

fkeujjpdc, 2017-03-15:
thank you,Use your method have solved the problem.

dpgeorge, 2017-03-15:
@pfalcon I think we can get rid of the part that tries to detect "-B" option.  It anyway doesn't work for make (ie doing "make -B" doesn't print the message `QSTR Looks like -B used, trying to emulate`).

pfalcon, 2017-03-15:
@dpgeorge : Ok.

dpgeorge, 2017-03-15:
Fixed in 23a693ec2d8c2a194f61482dc0e1adb070fb6ad4

dpgeorge, 2017-03-15:
The fix (removing -B detection) broke Travis: it seems that make on Travis behaves differently to the make I'm using locally, and if you call "make -B" then $? may be empty.  So I guess the fix mentioned above needs to be reverted, although I'd rather find a different solution to keep the line length of the command not too long.

dhylands, 2017-03-15:
If you can use $(findstring B,$(MAKEFLAGS)) then you can split the if into 2 statements (rather than if/then/else which is all one statment), one that tests to see if B is present and one that tests to see if it isn't.

dpgeorge, 2017-05-31:
Fixed by the #3108

closed 2017-05-31 by dpgeorge