micropython/micropython #2961: compile error in cygwin

fkeujjpdc, 2017-03-16:
16062501@rd-15070201 /usr/micropython-master/micropython-master/examples/embeddi
ng
$ make V=1
cc -std=c99 -I. -I../.. -DNO_QSTR   -c -o hello-embed.o hello-embed.c
process_begin: CreateProcess(NULL, cc -std=c99 -I. -I../.. -DNO_QSTR -c -o hello
-embed.o hello-embed.c, ...) failed.
make (e=2): 系统找不到指定的文件(The system cannot find the specified file)。
make: *** [hello-embed.o] 错误 2(Error 2)

system:Win7 ultimate x86
gcc version: 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496] (GN
U Tools for ARM Embedded Processors)
python: using install in windwos Python 3.6.1rc1

fkeujjpdc, 2017-03-16:
................
../mpy-cross/mpy-cross -o build-PYBV10/frozen_mpy/lcd160cr.mpy -s lcd160cr.py  m
odules/lcd160cr.py
process_begin: CreateProcess(NULL, ../mpy-cross/mpy-cross -o build-PYBV10/frozen
_mpy/lcd160cr.mpy -s lcd160cr.py modules/lcd160cr.py, ...) failed.
make (e=2): 系统找不到指定的文件。(The system cannot find the specified file)。
make: *** [build-PYBV10/frozen_mpy/lcd160cr.mpy] 错误 2(Error 2)

16062501@rd-15070201 /usr/micropython-master/micropython-master/stmhal

this is compile stmhal error
i use command "make V=1" compile

dpgeorge, 2017-03-16:
For the first error, you apparently don't have the `cc` command available.

For the second error, you need to do `make -C mpy-cross` first (from the root directory of the repository).

fkeujjpdc, 2017-03-16:
thanks for your help,after i compile mpy-cross by gcc, then i compile stmhal,i occur this error:
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
MPY modules/lcd160cr.py
Traceback (most recent call last):
  File "modules/lcd160cr.py", line 1
SyntaxError: invalid syntax
make: *** [../py/mkrules.mk:113：build-PYBV10/frozen_mpy/lcd160cr.mpy] 错误 1

fkeujjpdc, 2017-03-16:
i'm sorry i have so many qustion,because my linux compile knowledge is very poor. is there have micropython designed document ,i can learn it by myself, can you give me  a web link or a book.i think this project is very greate,i'm interested in this project , i hope learn it deeply

dpgeorge, 2017-03-16:
> thanks for your help,after i compile mpy-cross by gcc, then i compile stmhal,it occur this error:

It looks like the problem is that you don't have native symlinks working under cygwin, and so the file modules/lcd160cr.py is just a single line containing the location of the true file.  Therefore mpy-cross gives a syntax error (because it's not a valid Python script).

To work around this problem, either delete all the files in stmhal/modules/ directory (they are just symlinks); or copy the true files from drivers/display/ to stmhal/modules/.

Note that you might come across more problems using cygwin.  I suggest you install a virtual machine (eg virtualbox) and run a virtual Linux installation (eg Ubuntu).  This would be the easiest way to solve all your problems and prevent new ones from coming up.

And in the future, questions like these that are more general help questions than bug reports should be asked on the https://forum.micropython.org.

closed 2017-03-16 by dpgeorge