micropython/micropython #2964: Module struct not available

hanneseilers, 2017-03-17:
Compiled latest master branch version. But it's not possible to import module struct
    
    >>> import struct
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    ImportError: no module named 'struct'

peterhinch, 2017-03-17:
Try ustruct:
```python
import ustruct as struct
```
MicroPython has micro libraries, typically subsets of the Python library and with names prefixed with 'u', designed for small size.
Queries like this are best raised in the forum rather than in the bug tracker.

pfalcon, 2017-03-17:
More information in the docs: http://docs.micropython.org/en/latest/pyboard/library/index.html#python-standard-libraries-and-micro-libraries

closed 2017-03-17 by pfalcon