micropython/micropython #2969: py: Confusing error message when function passed a keyword arg that's not expected

dpgeorge, 2017-03-21:
Consider this code snippet:
```python
def f(x):
    pass
f(y=1)
```
Errors reported are:
- In CPython: `TypeError: f() got an unexpected keyword argument 'y'`
- In uPy: `TypeError: function does not take keyword arguments`

I hit this when calling a user-defined function and was very confused because all user-defined functions accept keyword arguments (only built-in ones may not accept kw args).  IMO the error message should at least be changed to be less confusing, but if we can spare bytes it should report which particular kw arg is causing the problem (and that's quite a simple change since at that point in the code it is known what the kw is that couldn't be found).

peterhinch, 2017-03-21:
See also issue #1992

dpgeorge, 2017-03-22:
Fixed by 58f23def55a705357b352ce8af642e03cd5c49f0, the offending kwarg is now reported (the patch actually decreased code size, or kept it the same).

closed 2017-03-22 by dpgeorge